bigtreetech/BIGTREETECH-TouchScreenFirmware #2021: Issue with [PR] Clickable upper icons during print #1996

Wombat37, 2021-06-23:
This is a very minor point but it totally confused me initially and is inconsistent with the idle screen operation.

### Description
A nice new feature introduced with PR #1996 is to enable quick access to some of the print settings during an active print. It achieves this by allowing the upper icons to be clicked upon to enable direct access to the controls that adjust these settings - by cutting down on the number of key presses needed to access those controls.

While most of the icons work as intended to access the controls, adjustment of the T0 and Bed temperatures on my system require extra keys presses. This is because when the T0 or Bed icons are clicked on, the state of the T0/Bed toggle icon on the control page is indeterminate - I think it's left in whatever setting was used last. This means that in some instances, an extra key click is required before the temperature can be adjusted, which rather undermines the whole point of the enhancement in the first place.

To illustrate the issue better, here are some screen shots:

[image]
Screen shot of main screen during active printing

[image]
Screen after clicking on T0 icon - working as expected

[image]
Screen shot after returning to main screen after clicking on Back button and pressing Bed icon. Nozzle temperature adjustment still showing.

The Nozzle icon now has to be clicked on to select the Bed icon before the Bed temperature can be adjusted.

Now when I go back to the Main screen and click on the Nozzle icon again, this time, it brings up the Bed temperature adjustment.

Not a big deal but creates a lot of messing around and probability of adjusting the wrong temperature zone - for an enhancement that's meant to cut down on key clicks.

For reference, when the printer is idle (Ready), clicking on the T0 con, brings up the Nozzle temperature control and clicking on the Bed icon brings up the Bed temperature control. I feel the same should occur during active printing instead of leaving it apparently random.

### Hardware Variant
BTT-E3-RRF_V1.1
BTT-TFT35-v3.0

### TFT Firmware Version & Main Board Firmware details
RRF 3.3_2
E3.27.x (latest as of today)

radek8, 2021-06-23:
read https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2015

kisslorand, 2021-06-23:
Those two icons are in fact one single button that calls the Heat menu.
Without the icon clickable option you would have to click More and than Heat icon that would bring up the same Heat menu.
I never heard before that Heat menu would create "a lot of messing around and a strong probability of adjusting the wrong temperature zone".

>...for an enhancement that's meant to cut down on key clicks and make things easier for the user

Those two icons do exactly that, cuts down on key clicks, they get you directly to the Heat menu, nothing more, nothing less.

The Speed/Flow menu is the same, yet no one has a problem with it...

Anyway, this is not a bug, it's by design, it's intended to be this way.

radek8, 2021-06-23:
@kisslorand,  I know you wanted to keep the 8 buttons.
But wouldn't it be better to make an exception on this screen?
if 10 buttons were defined, it would be better
the button would be split into a nozzle and a bed, and another button could toggle the height of the layer with the vest number in the future, as you did with the time and percentage of progress.

[image]

kisslorand, 2021-06-23:
It's absolutely possible.
My main concern is with more than 8 buttons is that other developers will start nagging me again.
Get their approval and it will be done in no time.

radek8, 2021-06-23:
@guruathwal , @digant73 , @bigtreetech ...
Do you agree to have an exceptional 10 buttons on the print status screen? It would be a use for them and it is a shame not to use it.

[image]

Wombat37, 2021-06-23:
I didn't mean to cause all this anguish. Clearly I didn't realize that during active printing that the T0 and Bed buttons were actually the same whereas on the Ready screen they are independent. Seems a bit weird to develop them differently but there it is.

This is a minor issue and I can easily live with the new implementation - however, it really did cause me initial confusion when both set points were off (i.e. set to 0) and it wasn't obvious which one I was adjusting.

Good luck with this.

oldman4U, 2021-06-23:
@kisslorand So far everything you made new has been merged. And this would be new.

guruathwal, 2021-06-23:
No one has provided any objection to PR #1996. I do not know why but there is a trend emerging now to treat opinions as transgressing, which is really sad for this community. I have now decided not to offer my useless opinion/feedbacks to avoiding this.

@radek because you asked specifically, IMHO, the 8 icons thing was never the reason for the printing screen not having clickable status icons when I implemented the new printing screen. The current implementation of clickable status icons after #1996 is unjustified. Smaller icons (Fan and Speed) are clickable separately but hotend and bed icons are club together which is clearly confusing and inconsistent, and for the reason being implied disapproval of the people who were not even involved in the discussions relating to #1996.

radek8, 2021-06-23:
@guruathwal,
Thank you for your comment. Yes, merging icons is confusing and the errors reported by the user are proof of that.

So Kisslorand can you define the buttons according to my design?
If someone else has an opinion, other alternatives may be discussed over your PR. :-)
I would do it myself, but unfortunately my programming knowledge is not enough.
Thank you

kisslorand, 2021-06-23:
I jumped too fast to say it's easy to implement 10 buttons on a page...
Everything is written for 8 buttons, page draw, icons draw, clear gaps on menu refresh, etc. It will increase code more than I anticipated, but hey, you asked for it.

radek8, 2021-06-23:
I'm not forcing you, pokuid it's against your conscience.

kisslorand, 2021-06-23:
"pokuid" ?

radek8, 2021-06-23:
fixed

kisslorand, 2021-06-23:
It is not about my conscience. 
I am torn between user experience and the code behind it,

Wombat37, 2021-06-23:
I'm sorry I made your life hell by being confused. 

I will try to do better in future :)

kisslorand, 2021-06-24:
Check PR #2026.

PR #2026 (2021-06-28): Extended clickable icons in printing menu
### Requirements

BTT or MKS TFT

### Description

This PR extends the clickable icons function in the printing menu. Each upper status icon has it's own function, except the Layer icon, which has no function associated yet, albeit it is clickable.

### Benefits

Nozzle status click brings up the Heat menu with the current nozzle preselected, clicking the bed status icon brings up the Heat menu with the bed preselected . Hopefully this clears the confusion created by previous implementation where those two icons had the same functionality, before they  [... cut]
closed 2021-06-28 by bigtreetech