bigtreetech/BIGTREETECH-TouchScreenFirmware #2023: FADE HEIGHT

3dfixer, 2021-06-23:
If i set fade height to zero, my prints come out the right height.
Now my bed is quite level, within 0.18 mm over the whole bed. if i set fade to say 10mm, my prints come out
1.3mm too tall up to this 10mm height, then the rest of the height is right, albeit 1.3 higher
i.e  at 3mm - 0.39 too high, 5mm - 0.65 too high, 10mm - 1.3 too high, 50mm - 1.3 too high, 98mm - 1.3 too high
the only thing i can think of is my probe height is -1.35

Question is - is this tft or a marlin problem?
And has anybody else noticed this?

btt skr 1.4  tft 35 v3.0 build as of 13/6/21, marlin 2.0.8.2
although had problem with older releases

digant73, 2021-06-23:
It is on Marlin only. TFT simply allows to set the fade value

radek8, 2021-06-23:
If you updated Marlin before this problem, initialize Eeprom again.
M502
M500
reset

oldman4U, 2021-06-26:
I have not noticed this. So when I print a 10mm object with the same fade hight it should be higher (probably for the amount of my probe offset)?

radek8, 2021-06-26:
> To jsem si nevšiml. Takže když vytisknu 10 mm objekt se stejnou výškou vyblednutí, měl by být vyšší (pravděpodobně pro část mého odsazení sondy)?

Absolutely not!

oldman4U, 2021-06-26:
Just printed a 2mm display cover with a z fade hight of 10 and the print came out 1.98mm.

Two options I see for you. Reduce the fade hight which shouldn't be a problem for you with a bed like this or spend hours to find the problem which is unlikely TFT related.

Happy printing 

Please help the community and close the ticket once you do not need it anymore. Thank you

3dfixer, 2021-06-30:
sorry i haven't been here to respond, work is nuts.
i thought it was a marlin thing so ill take it up there.
was hoping more people had same issue

closed 2021-06-30 by 3dfixer