bigtreetech/BIGTREETECH-TouchScreenFirmware #2024: [BUG] Extruder doesn't lower after multiple runout detections (Bug)

gordo3di, 2021-06-24:
### Description
If the runout sensor triggers but you don't put in filament or it false triggers again immediately, the buffer will keep filling up with the runout raise height and it will not lower back down to the same height. We have ours set to raise 10mm so with each trigger it will add 10mm. Additionally, we babystepped the extruder down (in this case 20mm) to continue the print however when it reached the next layer it tried to lower an additional 20mm into the model causing sad faces all around.

### Steps to reproduce
1. Trigger a runout
2. Don't insert filament
3. Watch extruder raise each time

**Expected behavior**
1. Runout triggers and extruder goes to parked position and raises up.
2. If another runout is detected, the extruder should lower back to the original height before raising again or it should ignore the raising.

### Hardware Variant
SKR v2 rev b
TFT35 V3
Marlin 2.0.8.2
TFT firmware from June 14, 2021

### Additional Information
I'll try to find the code and fix myself and post here if I'm successful unless someone else knows the code better and can do an easy fix. I can't remember seeing this issue posted here yet.

oldman4U, 2021-06-25:
Which sensor?

Watch extruder raise each time - each time what happens. After runout you have to press at least Resume...?

gordo3di, 2021-06-25:
It's our own sensor using an optical endstop not a smart sensor. The runout works fine unless you forget to insert filament and a runout triggers a second time.

Once a runout triggers, the extruder raises and goes to the parked position and it says Filament runout on the screen. I hit Ok and the Z is now at 10.02mm (the layer height plus the raise height).

If I don't insert filament and hit resume the runout will immediately trigger (Filament runout hit ok etc..), the extruder will go back to the spot it was on the print. Then it will go back to the parked position only now it's another 10mm higher. So now the Z is at 20.02mm. 

It will add 10mm each time a runout occurs and the print height is off.

So basically if someone forgets to insert filament on a runout or the sensor is failing they just lost their print.

oldman4U, 2021-06-25:
I can confirm this behaviour.

oldman4U, 2021-06-25:
Hope to have a fix for you by tomorrow.

gordo3di, 2021-06-25:
> Hope to have a fix for you by tomorrow.

That's awesome thank you. I was playing with the runout initializing code and starting to make a mess of everything.

oldman4U, 2021-06-25:
This is why I wanted to let you know. Help is on the way😁

closed 2021-06-29 by bigtreetech

gordo3di, 2021-06-29:
I finally had a chance to test this but I still have the same behavior. Still triggers a runout and adds 10mm (my raise height) each time if i don't insert filament. @oldman4U did you get to try this?

oldman4U, 2021-06-29:
Works for me. You tested PR or master?

gordo3di, 2021-06-29:
Tested master from this morning after bigtreetech merged it.

kisslorand, 2021-06-29:
Yes, it makes a difference. Where do you print from?
I can trigger this behaviour only if filament runout sensor is connected to the TFT and printing from onboard SD. I couldn't trigger this behaviour in any other situation.

gordo3di, 2021-06-29:
Yes we print from SD usually and runout is connected to tft. We're trying to work out all the situations that our customers might come across and limit confusion.

 Should we switch it so the runout is connected to the skr instead?

kisslorand, 2021-06-29:
Which SD? The on on the TFT or the one on the motherboard?
As I said I couldn't trigger this behaviour if the filament runout sensor was connected to the motherboard or if I tested from any other source than onboard SD.

gordo3di, 2021-06-29:
Sorry. We print from SD on the TFT and we expect most of our customers will as well. Early in my testing I think there were issues with the TFT and SKR regarding the runout if it was connected to the SKR so I instead connected it to the TFT.

oldman4U, 2021-06-29:
I have been testing connected to TFT and printing from Udisk.

oldman4U, 2021-06-29:
I use emulated m600 btw

gordo3di, 2021-06-29:
> I use emulated m600 btw

I turned that on as well. I just modified the thermal runaway code to trigger a runout via M600 instead of killing the printer and I think the emulated M600 works fine.

oldman4U, 2021-06-29:
Will check the master asap and let u know.

oldman4U, 2021-06-29:
It works...

https://user-images.githubusercontent.com/54359396/123867914-476dca80-d92f-11eb-93d7-17e1097b4cc1.mov

https://user-images.githubusercontent.com/54359396/123868025-72581e80-d92f-11eb-8e3c-c436efac5867.mov

https://user-images.githubusercontent.com/54359396/123868111-8b60cf80-d92f-11eb-8992-24476939884a.mov

gordo3di, 2021-06-30:
> It works...
> 
>  1st.mov 
>  2nd.mov 
>  3rd.mov

I can get the same behavior when printing from udisk but not when printing from the TFT sd card.

oldman4U, 2021-06-30:
Tested printing from TFT sd card - works. 

Could you attach a movie as well please.

gordo3di, 2021-07-01:
> Tested printing from TFT sd card - works.
> 
> Could you attach a movie as well please.

I just double checked it and it works perfectly now! I must have put on an older compiled version. Too many things happening over here. Thank you everyone for your help!

oldman4U, 2021-07-01:
🙌

PR #2035 (2021-06-29): Fix runout trigger during resume
- Fix runout trigger during resuming which is causing runout to trigger during resuming. (#2024)

Fixes #2024