bigtreetech/BIGTREETECH-TouchScreenFirmware #2025: [Q] Marlin UI with TFT35?

JohnnyTheOne, 2021-06-24:
Is it possible to use Marlin UI with TFT35?

oldman4U, 2021-06-25:
A TFT35 with or without encoder wheel?

JohnnyTheOne, 2021-06-25:
With v3

oldman4U, 2021-06-25:
TFT35 v3 has an encoder wheel and supports Marlin emulation mode.

How is this related to the TFT firmware?

JohnnyTheOne, 2021-06-25:
it's more about hardware that support MarlinUI touch mode.

oldman4U, 2021-06-25:
BTT TFT's support Marlin through encoder knob.

radek8, 2021-06-25:
> it's more about hardware that support MarlinUI touch mode.

does not support

JohnnyTheOne, 2021-06-25:
will it be in the future?

oldman4U, 2021-06-25:
In case a user is willing to spend his time developing it - yes.

What would be the benefit compared to the current interface?

JohnnyTheOne, 2021-06-26:
Most important configuration in one place

oldman4U, 2021-06-26:
So to be able to change a configuration in one place you need a Marlin interface with touch screen and not using an encoder knob?

Which settings and how often do you change them?

oldman4U, 2021-06-28:
Would you please be so kind and answer my questions above. Thank you

JohnnyTheOne, 2021-06-28:
In general it is better that the config is in one place. Marlin and BTT firmware are constantly updated, if my printer info changes every now and then I have to update firmware on both. if I change something a value I have to do it in two places that should not be necessary.
If BTT gets info from marlin without double config that would be nice. in marlin simulator and marlinUI it is.

`Would you please be so kind and answer my questions above. Thank you`
It was weekend

radek8, 2021-06-28:
But TFT is not stored by marlin parameters.
TFT reads Marlin parameters when turned on.

oldman4U, 2021-06-29:
Marlin is for the mainboard and this firmware is for the TFT. There are already some commands in use where the TFT reads parameters from Marlin and I am sure there will be more in the future.

The TFT firmware allows to configure all parameters in a single file using a simple text editor without the need to compile any software. Not sure there is an easier way to handle this. Have you asked for the same in the Marlin repository?

JohnnyTheOne, 2021-06-29:
I keep using marlin simulator with BTT TFT V3.0 without touch mode.

Thanks for your time and explanation.

oldman4U, 2021-06-29:
Please help the community and close the ticket once you do not need it anymore. Thank you

JohnnyTheOne, 2021-06-29:
It concerns the same parameter in two different files

radek8, 2021-06-29:
which parameter specifically?

radek8, 2021-06-29:
All printer parameters are stored on the Marlin board in EEPROM.
The TFT display reads them when the printer is turned on and can edit them.
When the printer is turned off, the TFT is forgotten. It does not have its own EEPROM for storing MARLIN parameters.
TFT only stores its own display parameters.
If you are missing an important parameter on the TFT, a programmer may try to make this parameter available in touch mode.

radek8, 2021-06-29:
If you edit a parameter in the simulated Marlin environment, the parameter also automatically changes in the touch mode display.
And vice versa.

JohnnyTheOne, 2021-06-30:
@radek8 I will ask my question differently. is it possible to use MarlinUI with BTT TFT V3 hardware? in addition to marlin simulator and BTT TFT firmware or instead. answer to this is no or maybe later. 

`which parameter specifically?`
All parameters
`If you are missing an important parameter on the TFT, a programmer may try to make this parameter available in touch mode.` 
I appreciate that

radek8, 2021-06-30:
The TFT display with this firmware does not support MarlinUI.
You would have to create a new FW with MarlinUI support

closed 2021-06-30 by JohnnyTheOne