bigtreetech/BIGTREETECH-TouchScreenFirmware #2027: [BUG] PrusaSlicer: "Filament Runout"-Error

omexlu, 2021-06-25:
Hello,

I have a strange problem with my Artillery Genius in connection with the PrusaSlicer, as soon as I want to start a print it shows me the error "Filament Runout", but this only happens when I have sliced with the PrusaSlicer (so I assume an error here, or a command that does not fit in the start code).

Attached is the G-Code file that was sliced with PrusaSlicer:

https://easyupload.io/ojbql3?fbclid=IwAR2sn1Xi7Wv354WgANiavz1fJZfGZq7e9Nt8flA1zgAhdsxXZhV3a_UmOCE

PS. Don't know if it has something to do with the TFT Firmware, but possible.

oldman4U, 2021-06-25:
Slice the same object in the two slicers and compare the gcode.

This repository is about possible bugs and feature requests related to the TFT firmware.

omexlu, 2021-06-26:
I have done it and compared not a big difference, another user tell me that he has this problem with cura, so very strange behavior, I can be a possible big in TFT firmware.

Normally the slicer should not do any big difference but here it's the case, it should be investigated.

omexlu, 2021-06-26:
Here is the same objects sliced with cura:
https://easyupload.io/7k323y

oldman4U, 2021-06-26:
The TFT sends the commands written in the gcode file to Marlin.

Do you have a filament sensor?

omexlu, 2021-06-26:
Yes, of course, otherwise it would be obvious where the problem lies.

But the error message comes right at the beginning and also only with prusaslicer at Cura not.

radek8, 2021-06-26:
Is it the same problem if you print from Marlins?

omexlu, 2021-06-26:
Dont havent try this i only peints Form screen over TFT screen,but only happends with prusaslicer.

oldman4U, 2021-06-26:
I have been testing both gcode files and they printed just fine without any problem.

Sorry that I ask, but the sensor is connected to the TFT?

omexlu, 2021-06-27:
Yes it's connected to TFT, unfortantly with me I get this weird error and dont know why it only happens with prusaslicer :/

omexlu, 2021-06-28:
I tried with the settings from cura (start / endcode) same result,disabling the filament sensor works for a few seconds then the screen freeze,there must be either a bug in the tft firmware or in prusaslicer (profio for genius).

oldman4U, 2021-06-28:
I can not reproduce the issue so I can not help.

Sorry

omexlu, 2021-06-28:
Thanks anyway, also do not know what this is :/

omexlu, 2021-06-28:
Check out my thread here, they are also assuming a firmware issue related to PrusaSlicer:
https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-filament-runout-error/#post-464743

oldman4U, 2021-07-01:
See #2040 maybe related

omexlu, 2021-07-01:
Yes seems so that its related, i cant use prusaslicer because this errors. Cura Works fine so i purged prusaslicer.

oldman4U, 2021-07-07:
Please close this ticket in case you are not looking for a solution anymore.

Happy printing

closed 2021-07-08 by omexlu