bigtreetech/BIGTREETECH-TouchScreenFirmware #2028: Error when making home with Z

Dark0nd0r, 2021-06-25:
When making home and going down Z sometimes he touches the Z switch, sometimes he doesn't touch it, as if he knows where it is, in the case of X and Y he always touches them.
This causes the height of Z to vary in the first layer, in order to do so I must increase the value of Z so that it compensates the height with the baby steps.
Thanks for your help.
(Sorry for my English)

Ender 3
Marlin 2.0.7.2

oldman4U, 2021-06-25:
Any hint that the problem is related to the TFT firmware?

You mention your printer and the Marlin version and no word about your TFT. This repository is about possible bugs and feature requests related to the TFT firmware.

Dark0nd0r, 2021-06-25:
> ¿Algún indicio de que el problema está relacionado con el firmware TFT?
> 
> Mencionas tu impresora y la versión Marlin y no se sabe nada sobre tu TFT. Este repositorio trata sobre posibles errores y solicitudes de funciones relacionadas con el firmware TFT.

My TFT is BIGTREETECH TFT35 E3 V3.0 Touch Screen 12864 LCD Display WIFI Module 3D Printer Parts For Ender3

oldman4U, 2021-06-26:
Same display here. 

While homing, the TFT sends the command to home to Marlin which does it. Use esp3d to home to see if it makes a difference. 

Because homing the axis is never 100% the same and the  accuracy of the z axis is critical the babystep function is available.

Dark0nd0r, 2021-06-27:
> esp3d

guruathwal, 2021-06-28:
This issue is not related to the TFT firmware. The TFT only sends a G28 gcode. All of the processing is done on the Mainboard. It is related to Marlin firmware or the printer hardware. It may also be related to the accuracy of your Z probe. You can ask the question in marlin forums or the manufacturer of the printer.

Dark0nd0r, 2021-06-28:
Thanks for the responses (sorry for the delay in responding).
I did the tests and I see that they are the same results for commands as for the screen.
The result is that he should touch the switch twice, sometimes he touches it the first time and the second time he does not, and when he does the home again he does it the other way around.
I am going to update Marlin and thus leave doubts.
I'll let you know when I do.

oldman4U, 2021-06-28:
The slower you home the more accurate it will be.

If the endstop clicks or not does not matter.

Z axis we speak about the fraction of a millimeter and using a probe instead of the endstop will make it more accurate.

Please close this ticket here because it is clearly not TFT firmware related.

Thank you

closed 2021-06-28 by Dark0nd0r