bigtreetech/BIGTREETECH-TouchScreenFirmware #2030: failed USB reading

klimma1508, 2021-06-26:
Hello, I have display Bigtreetech TFT35-E3 v3.0 and when I try to look what is on usb drive, then it write: "read U disk error", but this error it write after some time, when I bought it it worked great. Where can be problem? Thank you.

oldman4U, 2021-06-26:
This repository is about the firmware of the TFT. Did you change the firmware right before the problem started?

klimma1508, 2021-06-26:
No, I didn't change the firmware. I think that problem isn’t in the HW, because I got another display with this problem. It sees USB, but it can't read it.

oldman4U, 2021-06-26:
But you wrote that it worked before and now it does not work without changing the firmware. This repository is not a user group but dedicated to the firmware which runs the TFT displays.

kisslorand, 2021-06-26:
> Where can be problem? Thank you.

Probably your USB stick.

Dark0nd0r, 2021-06-27:
> ¿Dónde puede haber un problema? Gracias.

Your USB memory format it to FAT32, maybe this is the problem.

klimma1508, 2021-06-27:
Thank for help. I founded, that i need to use MBR and then format it to FAT32. I reformated usb on windows and here was mistake.

closed 2021-06-27 by klimma1508