bigtreetech/BIGTREETECH-TouchScreenFirmware #2031: [Q] Change temperature during heat.

tome9111991, 2021-06-26:
Hello 
I have a TFT35 E3 V3 with E3 turbo.

I would like to print a file that is preset for PETG. but would like to use PLA this time.

is in the Gcode
nozzle 240
bed 70

if I change the temperature during heating via the display (touch mode) like nozzle 200 bed 60 
they will not be taken over immediately.
 that means it first heats up to 240/70 makes the first layer then it only changes to 200/60

is there a way that the new temperatures are taken over immediately?

oldman4U, 2021-06-26:
Please read this and check your gcode after that: https://marlinfw.org/docs/gcode/M109.html

tome9111991, 2021-06-26:
> Please read this and check your gcode after that: https://marlinfw.org/docs/gcode/M109.html

Is not a Gcode issue.
With my Original display i had no issue with same gcode

kisslorand, 2021-06-26:
Where do you print from? (onboard SD, TFT SD, TFT USB)

tome9111991, 2021-06-26:
> Where do you print from? (onboard SD, TFT SD, TFT USB)

onboard SD

kisslorand, 2021-06-26:
Here's your problem. In your gCode you must have "M109 S240". That is a blocking command. Marlin will not take any commands until the nozzle reaches 240 degree Celsius. There's no way around it.
You can go round this if you print from TFT SD or TFT USB as the TFT will block M109 and will send M104, which is not a blocking command and you can override it.
BTW, why not just simply edit your gCode file???

tome9111991, 2021-06-26:
> Here's your problem. In your gCode you must have "M109 S240". That is a blocking command. Marlin will not take any commands until the nozzle reaches 240 degree Celsius. There's no way around it.
> You can go round this if you print from TFT SD or TFT USB as the TFT 
I can edit gcode well.
I am surprised because I didn't have this problem with the original display

kisslorand, 2021-06-26:
>I am surprised because I didn't have this problem with the original display

Allow me to doubt that.

tome9111991, 2021-06-26:
I will try with marlin mode.

Maybe the issue is TFT Port 🤷‍♂️

kisslorand, 2021-06-26:
I already told you what's the issue. listen, no one, I repeat, "NO ONE" can stop M109 and M190.

oldman4U, 2021-06-26:
In my first reply I showed you what the problem is. You refused to read. You refused to think and you refused to solve the problem by yourself. Now another user tells you what the problem is and you refuse to listen. I hope you do not refuse to close this ticket.

Thank you

tome9111991, 2021-06-27:
Btw in marlin mode Temperatur change work!

In tune menu i change from
Nozzle 240 to 200
Bed 70 to 60

Bed is heating to 60.
The nozzle jumps back to 240 i go again in tune menu and change back to 200 the printer start the print.

Is not working in touch mode!

closed 2021-06-27 by tome9111991

kisslorand, 2021-06-27:
Did you verify it with a thermometer or you just base you theory from the values displayed on the screen?

tome9111991, 2021-06-27:
i don't care about the temperature itself.

but when I put the bed at 60 degrees it doesn't raise any further