bigtreetech/BIGTREETECH-TouchScreenFirmware #2032: [FR] (block touch input while in idle mode)

gitmacer, 2021-06-27:
**Is your feature request related to a problem? Please describe.**
If you want to "wake up" the display it's pretty likely you accidentally click on a button and trigger an action. That's especially the case when you set idle brightness to 0%. 

**Describe the solution you'd like**
Add a "block input while in idle" option to the brightness options.

**Describe alternatives you've considered**
Block input while brigtness is 0%

**Additional context**

discip, 2021-07-24:
That's exactly what I was thinking of some time ago!
And it reminds me of my good old **SonyEricsson W880i**! 😃
Back in those days, when you were typing in a message and the display shut off, because you paused the typing, you had not to fear that anything was deleted or sent when pressing a key, because the first press of any key activated the screen and only after the screen was turned on again the keys worked as intended.

### So definitely a must have. 👍

digant73, 2021-07-25:
@oldman could you include this FR to the list?

discip, 2021-07-25:
@digant73 
I think you meant @oldman4U. 😅

oldman4U, 2021-07-25:
Once it is closed. Yes. Maybe someone will fix it before. 😁

kisslorand, 2021-07-25:
How should it work? Always? Only if dim is under a certain percent? Only if dim is set to 0%?
Should it be configurable? If yes, how?
I have this "feature" on another printer and it drivers me crazy. I dim the TFT only at 50% and I can see where I touch. Once it is dimmed I have to press twice in the same spot.

oldman4U, 2021-07-25:
I would see it as an optional feature as soon as dimmed is activated in the firmware. The dim value should not matter.

gitmacer, 2021-07-25:
> How should it work? Always? Only if dim is under a certain percent? Only if dim is set to 0%?
> Should it be configurable? If yes, how?
> I have this "feature" on another printer and it drivers me [...]

That's the reason why I suggested: "Add a "block input while in idle" option to the brightness options".

discip, 2021-07-25:
As @kisslorand stated, it only makes sense to block touch input, if the dim value is low enough to make the display unreadable. Else it would be annoying.

If this is not to much/overloaded it would be nice, if one could configure:
1. whether it is activated,
2. below which value it should kick in.

thanks in advance

gitmacer, 2021-07-25:
> As @kisslorand stated, it only makes sense to block touch input, if the dim value is low enough to make the display unreadable. Else it would be annoying.
>
> If this is not to much/overloaded it would [...]
"whether it is activated" so as suggested.

"below which value it should kick in" There is only normal/idle brightness. If you choose if it should disable input while in idle or not you automatically also set if your choose brightness is over/under that "max" brightness. So that option is redundend.

kisslorand, 2021-07-25:
So you basically want this feature even if the idle brightness lets you see what you hit/touch?

discip, 2021-07-25:
@gitmacer
You are right, the way I put it, the second option is redundant.

@kisslorand
Perfect.👍
> It should be ***automatically*** activated ***if idle is set to 0%*** and on ***other idle values*** to be ***optional***.

gitmacer, 2021-07-25:
> So you basically want this feature even if the idle brightness lets you see what you hit/touch?

Yes I want the option. Imagine it's set to 5% because you want to reduce light to a minimum but still want to see that the printer is powered. When you want to actually read the screen you don't want to trigger an action.

Also where would you set the border between visible and not visible? Is 5% still visible? Isn't that ambientlight dependent? Isn't that user dependent?

> I say it should be automatically activated if idle is set to 0% and on other idle values to be configurable.

No reason against that. 
Also not sure how the logic works. But if you implement the feature it should also auto disable if brightness is 100% even though a time is set.

kisslorand, 2021-07-26:
@gitmacer
I see the option in config.ini something like this:

## Block touch action when idle brightness is above 0%
# Options: 1= enabled, 0 = disabled
# Default: disabled

block_touch_on_idle: 0

For the rest, to activate it when idle brightness is set to 0% or deactivate it if idle brightness is set the same as non-idle brightness, it would be hard coded, not user selectable.

oldman4U, 2021-07-26:
Kisslorand. You are a user who wants to have the option for the final decision. Hard coding dependencies completely unnecessary takes away this ability and the next user will ask for the reason for the limitation.

0-No dimming, 1-dimming with activation tap, 2-dimming without activation tap. And the users are free to choose if they want the function and/or dimming at all, or not and this independent of the dimming level. So you can keep your micro dimming and another user can have zero brightness.

gitmacer, 2021-07-26:
@oldman4U
> @kisslorand. You are a user who wants to have the option for the final decision. Hard coding dependencies completely unnecessary takes away this ability and the next user will ask for the reason for the limitation.

I guess you are referring to his comment before he edited it?

> 0-No dimming, 1-dimming with activation tap, 2-dimming without activation tap. And the users are free to choose if they want the function and/or dimming at all, or not and this independent of the dimming level. So you can keep your micro dimming and another user can have zero brightness.

Are suggesting a 3-option solution? Can you tell me the benefit you see in having another way to disable dimming. Isn't disable idle by setting time to "never" or setting idle brightness to 100% enough? 

Currently I am still convinced that my suggested implementation is the most suitable one. Easy to implement/use, covers all use cases.

I did a quick photo edit to make it clear what I imagine. (Maybe someone can think of a better button name)
[image]

kisslorand, 2021-07-26:
My proposal was the same with the difference that when idle brightness is set to 0% than always have the touch actions disabled regardless of this settings shown in the picture above.

discip, 2021-07-26:
It's already on its way [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2073).

discip, 2021-07-26:
@gitmacer
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2073

I think this is ready to get closed. 😊

closed 2021-07-30 by bigtreetech