bigtreetech/BIGTREETECH-TouchScreenFirmware #2033: [FR] Version Revision Management

Wombat37, 2021-06-28:
I recently seem to have got involved in the reporting and testing of a few issues with this firmware - and I am pleased to help.

However, now my PC is filling up with different versions of this firmware, some released some in development - all with the same file name!

Can we somehow identify these different versions? I suggest we include the commit number in the file names. Or is there some other way the developers are able to tell the version of a particular firmware build by looking at it?

radek8, 2021-06-28:
https://docs.github.com/en/desktop/installing-and-configuring-github-desktop/installing-and-authenticating-to-github-desktop
Use GitHub desktop.

radek8, 2021-06-28:
You'll only have one copy. You always see what PR it contains.

Wombat37, 2021-06-28:
OK - thanks for this.

I thought there must be a way!

I'll close this issue.

closed 2021-06-28 by Wombat37

radek8, 2021-06-28:
https://desktop.github.com/