bigtreetech/BIGTREETECH-TouchScreenFirmware #2036: I think I found a bug[Q] (Your Question?)

jbodhorn, 2021-06-29:
Om compiling marlin 2.0.9.1 for my skr mini e3 v2.0 + tft35 e3 v3 in touch mode when I go to adjust the sensitivity of the x axis in my ender 3 the menu shows x and x2 instead of x and y. I can't seem to adjust the sensitivity of the x axis, only the y axis which seems to show up as the x axis in the menu on the touchscreen 

Here are my config files for marlin and the tft
Marlin-2.0.9.1 Configuration.h
https://pastebin.com/HGEqbnAB
Marlin-2.0.9.1-Configuration_adv.h
https://pastebin.com/9XhSGued
Tft35 e3 v3 config.ini
https://pastebin.com/z8TAFfVb

radek8, 2021-06-29:
can you attach a photo?

oldman4U, 2021-06-30:
Yes, you found and a PR to fix it is on the way. #2039

alex926ru, 2021-06-30:
I noticed this problem too. And I solved it by fixing two files: (interfaceCmd.c) and (parseACK.c).
In the file (interfaceCmd.c) I wrote the case 914 section like this:
************************************************* **************
case 914: // parse and store TMC Bump sensitivity values
        {
          uint8_t i = (cmd_seen ('I'))? cmd_value (): 0;
          if (cmd_seen ('X')) setParameter (P_BUMPSENSITIVITY, STEPPER_INDEX_X + i, cmd_value ());
          if (cmd_seen ('Y')) setParameter (P_BUMPSENSITIVITY, STEPPER_INDEX_Y + i, cmd_value ());
          if (cmd_seen ('Z')) setParameter (P_BUMPSENSITIVITY, STEPPER_INDEX_Z + i, cmd_value ());
          break;
        }
************************************************* *************
and in the file (parseACK.c) I wrote the else if (ack_seen ("M914 X")) section like this:
************************************************* ************
else if (ack_seen ("M914"))
      {
        uint8_t i = (ack_seen ("I"))? ack_value (): 0;
        if (ack_seen ("X")) setParameter (P_BUMPSENSITIVITY, STEPPER_INDEX_X + i, ack_value ());
        if (ack_seen ("Y")) setParameter (P_BUMPSENSITIVITY, STEPPER_INDEX_Y + i, ack_value ());
        if (ack_seen ("Z")) setParameter (P_BUMPSENSITIVITY, STEPPER_INDEX_Z + i, ack_value ());
      }
************************************************* ***********
Now everything seems to work.
Please check the correctness of my corrections.

jbodhorn, 2021-07-01:
> I noticed this problem too. And I solved it by fixing two files: (interfaceCmd.c) and (parseACK.c).
> In the file (interfaceCmd.c) I wrote the case 914 section like this:
>
> case 914: // parse and store [...]

How did you trace the issue to those 2 files and how did you know how to edit them to fix the issue? I'm still new to 3d printing and don't know any programming, but today I was reading about marlin and read that it's coded in C++. ATM I'm assuming that is C++ code you edited and you used a debug feature to find the issue. Most all of the compiling I've done was automated or I was following a tutorial that had all the values the configs needed changed.

Since I wasn't getting any errors in Visual Studio code, it wasn't pointing me in the general direction of where to look for the issue I was having. I still haven't learned how to use the debugging functions in Visual Studio Code, but since I don't know C++ I dunno how much it would have helped anyway. I think I need some tutorials for C++ and on Marlin if you know of any that are good links would be awesome.

alex926ru, 2021-07-01:
> How did you trace the issue to those 2 files and how did you know how to edit them to fix the issue? I'm still new to 3d printing and don't know any programming, but today I was reading about marlin [...]

Good day! When compiled in the Visual Studio PlatformIO environment, these errors are not detected. Since this is a purely logical error, that is, an algorithm error. You can only calculate such an error by analyzing the code. And the Visual Studio environment is just an editor that allows you to conveniently work with code. I compared the new firmware with the one on which this error did not exist and a little logic allowed us to reveal these purely mechanical errors of the programmer. I myself am not an expert in C ++ programming.

PR #2039 (2021-07-01): Fix incorrect M914 parsing
- Fix incorrect M914 parsing due to wrong index. (#2036)
- Add `TODO` label to stale bot ignore list.
- 
Fixes #2036
closed 2021-07-01 by bigtreetech