bigtreetech/BIGTREETECH-TouchScreenFirmware #2037: BTT SKR e3 mini v2.0 and BTT TFT24 v1.1

Vilunki15, 2021-06-29:
Hi! I'd want to buy BTT SKR e3 mini v2.0 and BTT TFT24 v1.1. But I'm not sure to do they work with each other? I'd want to use marlin and TFT mode. 
[image]

oldman4U, 2021-06-29:
And you have access to youtube and Facebook?

oldman4U, 2021-06-29:
BTW, the TFT24 does not really fit the Ender 3 but the TFT35 E3 v3 is a 1:1 replacement and has a illuminated encoder and many features more.

Vilunki15, 2021-06-29:
> And you have access to youtube and Facebook?

Yes!

Vilunki15, 2021-06-29:
> BTW, the TFT24 does not really fit the Ender 3 but the TFT35 E3 v3 is a 1:1 replacement and has a illuminated encoder and many features more.

I know, but it's two times expensive. And I have already a case for that tft24.

oldman4U, 2021-06-29:
Then please check the options there.

This here is not a user group but a repository dedicated to the firmware of the TFT touchscreens offered by BTT. Once you got your BTT TFT, you can download free firmware for it from here and report bugs  and ask for new features. Does not matter if it is a BTT TFT24 or BTT TFT70.

It is a one time investment and the TFT24 is tiny and the quality of the display is worse compared to the much newer TFT35 E3. Your choice.

Please help the community and close this ticket. Thank you

radek8, 2021-06-30:
attention! A schming and a combination. The SKR Mini and TFT 24 1.1 have different connectors.

radek8, 2021-06-30:
You'd have to use a Y cable.

oldman4U, 2021-06-30:
@radek8 Do we expect users starting a new ticket to read The _Read me before posting a New Issue! #1047_. It is about the E3 mainboard and the encoder wheel changing the print speed in touch mode.

radek8, 2021-06-30:
@oldman4U 
I'm not talking about an encoder problem.
The TFT has EXP1 and EXP2 connectors
SKR E3 mini has an EXP3 connector
they cannot be interconnected without Y reduction.
It used to be in the readme on the front page, but you recently deleted it.

oldman4U, 2021-06-30:
I know you need a y cable. But I talk about the encoder wheel changing the print speed in case the TFT is connected to an E3 Mainboard.

When did I remove this information. This must have been by mistake😔

radek8, 2021-06-30:
This has been addressed several times
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/700

radek8, 2021-06-30:
Sorry, the info is still there, but I would think that there used to be a picture :)

radek8, 2021-06-30:
@Vilunki15 
In conclusion, I can say that I do not recommend this combination TFT and SKR

oldman4U, 2021-06-30:
I love my TFT35 E3 and SKR E3 DIP.

closed 2021-06-30 by Vilunki15