bigtreetech/BIGTREETECH-TouchScreenFirmware #2038: Configure BTT firmware TFT70 V3.0 v3.0.27

Decalis1979, 2021-06-29:
i have a Geeetech 3 hotend turned on, in version V3.0.26 the exhausts appeared normal on the screen (E0, E1 and E2). When I upgraded to V3.0.27, they no longer show up. I modified config.ini by putting hotend_count: 1 and (Extruder Count) ext_count: 3. And I can't move extruders E1 and E2 because they don't seem to select on the screen.

radek8, 2021-06-30:
Have you tried my advice?

Decalis1979, 2021-06-30:
In version xx.26 the mixer was already selected as you instructed me (#define MIXING_EXTRUDER 1) and it works perfectly. However, in this last firmware xx.27 that I went to update, the 3 extruders (E0, E1 and E2) do not appear on the display.

radek8, 2021-06-30:
In the precompiled FW, the default Mixing extruder is 0,
you must set it to 1, compile your own FW.
You must define the number of extruders in the cinfig.ini configuration file.

Decalis1979, 2021-06-30:
***** Version xx.27 is configured as below. However, the extruders (E0, E1 and E2) do not appear on the display:

#--------------------------------------------------------------------
# Printer / Machine Settings
#--------------------------------------------------------------------

#### Hotend Count
#   Value range: [min: 1, max: 6]
hotend_count:1

#### Heated Bed Support
#   Options: [enable: 1, disable: 0]
heated_bed:1

#### Heated Chamber Support
# The TFT will auto-detect if chamber heating is enabled in Marlin firmware.
#   Options: [enable: 1, disable: 0]
heated_chamber:1

#### Extruder Count
#   Value range: [min: 1, max: 6]
ext_count:3

#### Fan Count
#   Value range: [min: 1, max: 6]
fan_count:1

#### Controller Fan Support
#   Options: [enable: 1, disable: 0]
controller_fan:0

closed 2021-07-06 by Decalis1979