bigtreetech/BIGTREETECH-TouchScreenFirmware #2040: [BUG] Refresh printing page issue MKSTFT28

Help3D-Padova, 2021-07-01:
### Description

Menu navigation is fine. When you start a print (USB or SD) and try to click on icons during heating, the screen starts to lag till the temperature reach the target. If you try to stop the print the LCD freeze. While printing everything is ok.

### Steps to reproduce

1. Compiled on the 29th June last commit
2. Start a print from SD or USB
3. Touch the icons on printing menu

**Expected behavior**
Responsive click

**Actual behavior**
Lag, Refresh rate low

### Hardware Variant
MKS TFT28 - Artillery (not the last one) 

### TFT Firmware Version & Main Board Firmware details
TFT: Jun 29 commit
MKS: Marlin 2.0.7.2

Is something related to the actual Marlin Version ? Today i'll try the 2.0.9

digant73, 2021-07-01:
the TFT fw requires at least Marlin 2.0.8.1. I suggest to move to current version 2.0.9.1.

Help3D-Padova, 2021-07-01:
> the TFT fw requires at least Marlin 2.0.8.1. I suggest to move to current version 2.0.9.1.

Hi Antonino @digant73 

I tried your package attached to the thingverse item page and, even if is required 2.0.8.1, the FW is working correctly with the 2.0.7.2 onboard. 

Could be related to this: [bigtreetech#2026](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2026) or [this ](https://github.com/Help3D-Padova/BIGTREETECH-TouchScreenFirmware/commit/5b0744b8f85ebdb3199e2aa6f75194796de9d175)

digant73, 2021-07-01:
It's possible If you didn't have the lag/freeze pressing on the usual "stop" button while you got the lag/freeze pressing first on the new clickable icons (e.g. heat, fan etc...) and then pressing on the usual "stop" button. But I have to check on my side. I will try this night (I didn't update my printer to the last merged PRs yet)

Help3D-Padova, 2021-07-01:
also the filament sensor is not working correstly when M73 is in the Gcode. You can disable from the menu but when you start the print it trigger the filament sensor alert. 

Same gcode without M73 is working

digant73, 2021-07-01:
> @digant73 bug spotted, this issue appears ONLY on gcode with M73 . Without the time estimation is working fine.

Provide the gcode file you used so I will test it.
LCD_SET_PROGRESS_MANUALLY is not required on config.ini (and also on Marlin fw) printing from TFT's SD/USB because M73 is not sent to Marlin (M73 is entirely handled by the TFT). It must be enabled in Marlin fw only if printing from onboard SD (but in that case the TFT will not display any remaining time because Marlin doesn't provide that info to the TFT).
If you used my TFT fw from thingiverse and it was working with the same gcode file, then the problem is in some new PRs recently merged (my current TFT fw on thingiverse includes M73 support and it was properly working).

Help3D-Padova, 2021-07-01:
**PARTIAL SOLUTION**

**The issue is related to the USB PENDRIVE and Prusa Slicer.** 
SAME Gcode uploaded to a different USB Drive is working perfectly.

I was using the USB Drive included with Artillery, i switched to a spare Sandisk and no problem with the screen lag.

Filament sensor is still triggered random even with both usb drive or using IDM or Pslicer

Help3D-Padova, 2021-07-01:
@digant73 i found that was USB drive because i extracted the USB from the LCD during the heating of a print and i got a perfect refresh of the LCD

digant73, 2021-07-01:
ok, in any case attach your gcode file so I will verify if the problem on filament sensor is present on my printer

Help3D-Padova, 2021-07-01:
> ok, in any case attach your gcode file so I will verify if the problem on filament sensor is present on my printer

Ok tomorrow i'll attach gcode. 

# NOTE for users which have a filament sensor connected to the mainboard:
# Enable the start and end gcode below, and add the following commands to the start/end gcode: 
# start_gcode: M75\n
# end_gcode: M77\n

I'm using M75/M77 on my profiles (all slicers). I know that Marlin needs a M75 to know if is printing and then runout filament can be triggered. Now i have the Auto start for the timer, i have to try without M75 on the start gcode, maybe could be the issue of the random triggered filament runout at start ? i don't think but i've to test it out tomorrow  @digant73

omexlu, 2021-07-01:
I dont have the filament Sensor connecter to the mainboard and i have similar Problems (just with pursaslicer) i dont think that the Final solution and the USB Drive Is the Problem :)

Help3D-Padova, 2021-07-02:
@omexlu the USBDRIVE is only related to the slow response of the screen when using gcodes generated with Prusa Slicer.

The Runout filament sensor trigger also with IDM or S3D gcodes. The strange fact @digant73 is that i turn on the printer ----> Start a print from USB ---> Runout triggered immediatly . I stop the print and restart the SAME gcode non runout triggered :)

oldman4U, 2021-07-05:
Whats the status of this ticket?

digant73, 2021-07-05:
I don't have that problem on runout filament. I would suggest to try this setting"fil_noise_threshold:200"

oldman4U, 2021-07-05:
Was not able to reproduce it too.

Help3D-Padova, 2021-07-05:
@digant73 right now i'm out of office for holidays, i'll try asap with 3 other artillery to figure out if is FW related or HW related. 

I have to try another wall plug. Noise threshold tested up to 1800, better situation but not solved. With 100 it triggers also after the first runout. With 1000 or greater it triggers ONLY starting a print, after clicking OK the print continues correctly.

I'm using the V4.0 FW, i'll test the 3.0 but not the new one because if for the new HW in the latest Artillery produced, right ?

I'll keep you updated

digant73, 2021-07-06:
you can try the Genius fw version (it's derived from the v4 version) on the new ASX1/Genius models.

Help3D-Padova, 2021-07-08:
> Any news?

Updates, i'm using the latest commits.

MKS_TFT28_V3_0 - Not working, white screen
MKS_TFT28_V4_0 - Runout trigger
MKS_TFT28_NEW_GENIUS - Runout trigger

Tried also the @digant73 new package (1.27.x Patch 2, July 05 2021) same issue (filament Runout)

I tested on 3 different Artillery (with different MKS FW) and i have the same issue, tried to change the wall plug issue is still there. 
Tested different usb drive, gcodes (old and new) and slicers.

@digant73 using your package1.27.x Patch 1, June 02 2021 no issue at all, everything is working as expected

omexlu, 2021-07-08:
Same as me with prusaslicer (runout error) and freeze after stopping when errors occurs.

kisslorand, 2021-07-08:
Is filament runout sensor disabled on TFT's menu?

Help3D-Padova, 2021-07-08:
> I had already tested this some time ago, even if I deactivate it no longer comes to the runout error, but it still does not print and hangs up.
> 
> Furthermore, it is not the point to deactivate the filament sensor just to get something running, so the problem is somewhere else.
> 
> Some code in PrusaSlicer must trigger the problem here, no problem with Cura.

in my case is not related to the slicer, it triggers with Pslicer, IDM or S3D.

omexlu, 2021-07-08:
Then maybe several slicers, for me its only Cura thats working correct or not trigger the error :/

kisslorand, 2021-07-08:
> Furthermore, it is not the point to deactivate the filament sensor just to get something running, so the problem is somewhere else.

I know exactly why I asked that. If the filament runout sensor is not connected to the TFT than enabling of the TFT's filament runout sensor is pointless, more than that it can trigger false alarms. If the filament runout sensor is connected to the motherboard than the TFT relies on Marlin to be informed about the filament runout sensor's state, not the input port of its MCU.
   It would be great to know if it is Marlin related or TFT related. Keeping the TFT's filament runout sensor enabled whilst using it connected to the motherboard makes it harder to narrow the source of the issue.

omexlu, 2021-07-08:
Sure I understand, the filament sensor is connected to the TFT in my case.

digant73, 2021-07-09:
@Help3D-Padova please provide a sample gcode file so I will test it with my fw releases

digant73, 2021-07-09:
verify if this version is working properly

[MKSTFT28.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6792856/MKSTFT28.zip)

digant73, 2021-07-09:
please verify if everything is ok with the following fw.

[MKSTFT28.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6793222/MKSTFT28.zip)

If so , the problems seems caused by PR #2035. In particular it seems that the problem is due to the inverted code below (although it is logically correct):

  if (!isPrinting() || isPaused())  // Check for No printing or printing paused before reading sensor
    return;
  if (!FIL_IsRunout())              // Filament not runout yet, need constant scanning to filter interference
    return;

Restoring the old code below it seems the problem is not present:

  if (!FIL_IsRunout())              // Filament not runout yet, need constant scanning to filter interference
    return;
  if (!isPrinting() || isPaused())  // Check for No printing or printing paused before reading sensor
    return;

Help3D-Padova, 2021-07-10:
@digant73  "The bug is oneshot (only after a power on or reboot). Pressing on resume button or stopping and restarting a new print, there is no more false filament runout error message" 

yes but when you press resume the filament sensor is not working anymore, just try to pull out the filament :)

On Thursday late night i tested a build with the previous **extend.c** and it was working. I had to reply here in the post but i was out of office all day on Friday (and i needed more test to confirm). So yes the issue is related to this ! Next week when i come back from holidays i'll test more

omexlu, 2021-07-10:
With me unfortunately after resume all freeze :/

oldman4U, 2021-07-12:
Marlin #22327

digant73, 2021-07-14:
@omexlu please test PR #2061 (althought it seems you have another kind of problem) and let me know if it solves also your problem.
Furthermore, attach your gcode file causing the freeze, so I will test it

omexlu, 2021-07-14:
I can unfortunately no longer test because I have deleted the prusaslicer and only use Cura and stay with it.

So unfortunately I can no longer provide the G-code. I will stay with Cura here the problem is not present.

I'm sorry but unfortunately someone else must perform the tests whether it works with this PR now.

I had already uploaded the G-code in my bug-report but I don't know if the file is still available, but the files are still only in this post:

https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-filament-runout-error/#post-464701

digant73, 2021-07-14:
the prusa slicer file is no more available on that bug report

omexlu, 2021-07-14:
In the zip-file I remember I have included both files:
https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-filament-runout-error/#post-464701

Currently not at home to test.

digant73, 2021-07-14:
there is a zip file named 1624867158-SLICE_FILES.zip with the following two files:
AG_BLT_genius_holder_CURA.gcode
BLT_genius_holder_0.2mm_PLA_Genius_1h21m.gcodergere

Is the last one from prusaslicer?

omexlu, 2021-07-14:
Yes the Last ONE :)

digant73, 2021-07-14:
@omexlu both files are properly processed with no freeze by both the current fw and my PR #2061. Obviously on my PR, the runout bug is no more present.

omexlu, 2021-07-14:
Thanks for testing, good that the runout is solved, don't know why prusa dont work and freeze, maybe related to this bug.

omexlu, 2021-07-14:
No, because I tested it once with prusaslicer an immediatly haved this issue, after few tests I return to cura who have no problem with it, even with the filament bug in your pr, I don't have a issue here.

closed 2021-07-19 by bigtreetech

Sonicboom247, 2021-07-28:
why was this closed? is there an actual fix?.  I just tested again on a SWX1 with 2.0.9.1 and BTT 27 runout sensor on in TFT settings, M75 in start code and immediate runout alarm on start print. removed M75 and still runout alarm

oldman4U, 2021-07-28:
Ticket closed because the reported issue has been fixed in master firmware.

Sonicboom247, 2021-07-28:
> Ticket closed because the reported issue has been fixed in master firmware.

Thank you @oldman4U , are you referring to master in BTT firmware or Marlin 2.0.9.1 firmware? Downloaded latest BTT today and still get error. However my marlin is not bugfix.

oldman4U, 2021-07-28:
BTT.

Maybe you have another issue.

omexlu, 2021-07-28:
This is an issue that many people haves and is a issue within the TFT firmware.

Sonicboom247, 2021-07-28:
is it possible to reopen this issue as it is either still not fixed or the fix is not clearly defined.?  it seems to be more isolated to the MKS TFT 28 V4

PR #2061 (2021-07-19): bugfix #2040 + improved #2035 + minor cleanup
**BUGFIX:**
* fixed #2040. That bug was introduced by PR #2035. The FIL_IsRunout() function needs always to be periodically performed to maintain the proper runout state while PR #2035 avoided to perform FIL_IsRunout() in case of no printing or paused print. In case of runout, resuming the print forced FIL_IsRunout() to return (as initial state) always "false". That value will not allow to resume the print in case the runout sensor uses a not inverted logic (e.g. MKS TFT). In that case that initial value triggers a runout and the print is always fo [... cut]