bigtreetech/BIGTREETECH-TouchScreenFirmware #2041: Is G27 Working?

3dfixer, 2021-07-03:
is g27 implemented?
when i use it, the print stops, z raises, printhead moves to front, does not pause,  moves straight back to
print position, does not lower z, starts printing in thin air for 15/20secs then z lowers and off it goes with print.
question is  1- why does it not pause?  2- why start extruding before lowering z? 
i've tried this using both marlin/tft mode and both sd card slots, same thing happens

i want a pause to insert magnets

skr v1.4  tft35 v3.0

radek8, 2021-07-03:
G27 is a code that you must have activated and defined in Marlin.
It has nothing to do with  FW display.

oldman4U, 2021-07-05:
What is the status of this ticket?

oldman4U, 2021-07-07:
I can not find anything related to G27 in the TFT firmware. Why not using M600 and how could you add G27 to the print gcode?

3dfixer, 2021-07-07:
so it would seem like its a marlin thing, i will go chase it up there.
m600 is change filament, which is doable but not ideal.
and if you use prusa slicer g27 is a standard option for initiating a pause

closed 2021-07-07 by 3dfixer

oldman4U, 2021-07-07:
Using Cura I found no way to get g27 in the gCode. M0 or M600 was added using gCode scripts and I sent g27 during a print using the terminal instead. You could try Pronterface and see what it does.

3dfixer, 2021-07-07:
what happened when u sent g27?

oldman4U, 2021-07-08:
Same you described.