bigtreetech/BIGTREETECH-TouchScreenFirmware #2042: [FR] Option to forward all printer output unmodified to the additional UART port/s of the BTT TFT35-E3 V3.0

badbod, 2021-07-03:
**Is your feature request related to a problem? Please describe.**
All attempts to have any external control of the printer using UARTS 2,3,4 of the BTT TFT35-E3 V3.0 are broken as printer output is not relayed in full.

**Describe the solution you'd like**
An option to have all printer output relayed in full, without any changes, to the additional UART port/s of the BTT TFT35 E3 V3.0.

**Describe alternatives you've considered**
Maybe connecting the BTT TFT35-E3 V3.0 to controller board using ESP modules attached to TFT ports allowing simultanious access to all printer output by all wifi connected devices thus bypassing the BTT TFT35-E3 V3.0 altogether?

**Additional context**
Currently the BTT TFT35-E3 V3.0 additional UART's  serve no purpose. Only select printer output is relayed to them. This  renders any devices conncted to the additional UART ports unable to function in any meaningful manner..

radek8, 2021-07-04:
You have enabled the required ports in the Config.ini_ file

badbod, 2021-07-05:
yes,  the problem is, I see no ok or busy responses,   just temps and  some other bits, and some not coming from the printer like fr: and something else,   they are consistent and do not change with baud rate,    conneciting the same eauipment direct to the controlller board and I see all the printer responses at any baud rate.    I am sticking to 115200 while testing.    I see the same through ESP module connected to ESP port also,  ,    the ports are enabled, and only the ports in use, though I have enabled them all to test but no difference.    the temps and the other line I see,  and before it stalls the connection waiting ofr OK from printer,   like M115 and M420 do get meshes and are in full, not what I am used to seeing if the connection is iffy,  the ok andd bust etc that are missing are just completly missing.  

Is it supposed to be sending them through?  all the forum,s all over the net describe the same problem and thats why they cant use them for octorpint or anything else.  I think we all assume it is by design as the same data seems to be missing for eveyone

digant73, 2021-07-25:
Which is this fork implementing that?

oldman4U, 2021-07-31:
@badbod 

Are you watching your own ticket?

badbod, 2021-07-31:
HI,  no, I am not using that display any more.  Ill close this out.

closed 2021-07-31 by badbod