bigtreetech/BIGTREETECH-TouchScreenFirmware #2043: [M355 Caselight] dont work it after flash the new FW verson

rebel1985, 2021-07-04:
### Description

case light funktion it dont works, no on no off, it blink very short no percent regulation or else.
but with the light option menü the light are on and in differend colors.
maybee it is the M355 S0 or S1 Pxxx code....

### Steps to reproduce

1. flashed the firmware
2. wait for the init.
3. go to caslight and will control the percent of light
4. dont work it short blink and all neo rgb leds are off

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

Distribution date: 2021-05-15

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2021-07-04:
Update Marlin.
Check last TFT firmware which works.

rebel1985, 2021-07-04:
Marlin 2.0.8.1 with TFT Distribution date: 2020-05-12 it works

guruathwal, 2021-07-05:
@rebel1985 please close the issue if it is solved.

oldman4U, 2021-07-05:
Es gibt Abhängigkeiten und daher ist es wichtig eine Version zu benutzen wenn eine solche vorgegeben wird. Daher 2.0.8.2 oder 2.0.9.1 installieren sonst wird hier niemand helfen.

oldman4U, 2021-07-06:
Please help the community and close the ticket once you do not need it anymore. Thank you

rebel1985, 2021-07-07:
ja gut dann mache ich später mal ein FW abgleich auf die neue Marlin und dann passend die FW fürs TFT dazu

closed 2021-07-07 by rebel1985