bigtreetech/BIGTREETECH-TouchScreenFirmware #2045: TFT E3 V3.0 - Knob LED not working anymore

arminth, 2021-07-04:
I am compiling and flashing fimware on a regular basis to be up to date and well familiar with the procedures.

Today I realized, that my knob-LED was not working. I didn't pay attention in the past to it so I don't know since when it was the case. It definetly know that it already worked before. The rest of the display is working fine.

Could anyone check, if compiling the actual sources for a E3 V3 results in a working knob-LED? I have adjusted configuration.h and config.ini to set it to blue or white but its staying dark. Also changing it in the menu doesnt change anything.
If it works for anybody else with the most up to date sources, I might have a hardware problem with the display.
Any feedback bould be highly appreciated!

TIA!

Cheers
Armin

oldman4U, 2021-07-04:
Try v27 official release.

arminth, 2021-07-04:
I have to apologize.
I had a very old version which I tried and the knob worked. I have created a brand new branch, went through the entire configuration again, compiled and flashed. Now it's working again! I have no idea, whats wrong with my normal branch.

Sorry for bothering and thanks @oldman4U for looking at it!

cheers
Armin

closed 2021-07-04 by arminth

oldman4U, 2021-07-04:
Happy printing 

👍🏻