bigtreetech/BIGTREETECH-TouchScreenFirmware #2049: [Q] Bed temperature up to 350C

Vertorek, 2021-07-05:
Hello,

I am trying a broader maximum bed temperature up to 350C. When loaded a modified config file with 350C at the bed I received information about invalid parameters. The nozzle at 500C is fine in the case of the bed temperature 350C the maximum temperature without error about invalid data is 200C.

Is there something that would be necessary to change also?

in config file I changed:

#### Bed / Extruder / Chamber Maximum Temperatures
#   Format: [max_temp: T0:<max temp> T1:<max temp> T2:<max temp> T3:<max temp> T4:<max temp> T5:<max temp> BED:<max temp> CHAMBER:<max temp>]
#   Unit: [temperature in °C]
#   Value range: hotend:  [min: 20, max: 1000]
#                bed:     [min: 20, max: 1000]
#                chamber: [min: 20, max: 100]
max_temp:T0:500 T1:275 T2:275 T3:275 T4:275 T5:275 BED:350 CHAMBER:60

#### Preheat Temperatures
#   Format: [preheat_nameX:<name>]
#           [preheat_tempX:T<hotend temp> B<bed temp>]
#   Unit: [temperature in °C]
#   Value range: name:        [min: 3, max: 20 characters]
#                hotend temp: [min: 20, max: 1000]
#                bed temp:    [min: 20, max: 500]

oldman4U, 2021-07-06:
Just curious to know. What is the use case? Still 3D printing?

oldman4U, 2021-07-06:
BBQ?

I could imagine that there is a max temp for security reasons.

Vertorek, 2021-07-06:
Hi oldman4U,

High-temperature printer for PEEK, PEKK, PEI etc... 

General idea is to make this printer below 1000 EUR on easily accessible parts. Marlin/Klipper-based and open source. So printer in the range of 1 kg high-temperature semicrystalline filament.

BBQ probably I will try to make fry on it as well, what you say about steak I hope you are not vegan. 

I know that it was limited on purpose but I would like to deliberately omit this security feature do you have any idea how to do it?.

oldman4U, 2021-07-06:
All those filament start to soften around 200 degrees. So a heated bed with 350 would not only to br dangerous but would slso avoid that the filament keeps its shape and would melt on the bed.

I love BBQ but keep in mind that 350 is much to much for a good steak.

I do not know how to change it and would highly recommend to use zhe temperature which is possible. 

Please help the community and close the ticket once you do not need it anymore. Thank you

Vertorek, 2021-07-06:
I need something around 260C and your comments still not solving the problem. I realize how high I want to go and this is my choice.

A semicrystalline polymer such as PEEK and PEKK possess double melting behavior more you can read about it here: https://pubmed.ncbi.nlm.nih.gov/33673299/

Softening starts above 260C, glass transition temperature for PEEK is around 150C but that refers to the amorphous part of PEEK, crystallites are stable and their thermal resistance depends on printing parameters more in this paper.

Sonicboom247, 2021-07-31:
dangerous, my guess is "IF" the safety features are removed you would still need to either make sure your power supply can not only push the bed and nozzle to those temps but also maintain it for the entire print.  Also ensure the bed and anything touching it can handle those temps, as in, will the glue holding the manual bed leveling knobs hold ( if equipped). can the wiring handle it?  Honestly i would first test with the maximum settings allowed and see what happens. from there you will probably need to do some upgrading.

digant73, 2021-08-19:
@Vertorek check on Marlin fw (Configuration.h) the value for BED_MAXTEMP

Vertorek, 2021-08-21:
> @Vertorek check on Marlin fw (Configuration.h) the value for BED_MAXTEMP

BED_MAXTEMP in Marlin was/is set for 350C 

I can use pronterface without a screen and everything is fine but with a connected screen, there is a problem. It seems that the screen duplicates the thermal runaway safety feature I have to try with turning off, the thermal runaway for the bed but I would prefer to have it on.

digant73, 2021-08-22:
The max temp for bed configured in the TFT is 200. It is also reported in the description for max_temp.
#### Bed / Extruder / Chamber Maximum Temperatures
#   Format: [max_temp: T0:<max temp> T1:<max temp> T2:<max temp> T3:<max temp> T4:<max temp> T5:<max temp> BED:<max temp> CHAMBER:<max temp>]
#   Unit: [temperature in °C]
#   Value range: hotend:  [min: 20, max: 1000]
#                bed:     [min: 20, max: **200**]
#                chamber: [min: 20, max: 100]

if you want to increase that value you should change the following line of code in Config.h ans set the desired max value.

#define MAX_BED_TEMP            200  // bed temp over this will not pe parsed.