bigtreetech/BIGTREETECH-TouchScreenFirmware #2050: TFT24 v1.1 Bootloader request

Tomdein, 2021-07-05:
Because of low activity in BIGTREETECH-TFT24-V1.1 repo I am posting a duplicate request like in the previously mentioned repo ([BIGTREETECH-TFT24-V1.1 - issue #33](https://github.com/bigtreetech/BIGTREETECH-TFT24-V1.1/issues/33)).

Here is the copy of the text:
"My brother shorted SKR 1.4 and therefore also fried the connected BTT TFT24 v1.1.

We repaired SKR 1.4, flashed new bootloader and firmware.

Now we repaired TFT24 v1.1, but we are missing a bootloader.
Could you publish the bootloader for TFT24 v1.1 so we can flash it via the ST-Link v2?

Thanks."

Also if anyone could dump the the bootloader or the whole chip it would be great. I could work with that.

Thank you.

closed 2021-09-10 by stale[bot]

rbarreiros, 2022-09-16:
And another one needing the bootloader firmware.... bootloader is corrupted, need to reflash, anyone has it around ?

rbarreiros, 2022-10-02:
Any chance at all of someone that has a tft24 and either a usb/uart adapter or stlink that can get the bootloader ?