bigtreetech/BIGTREETECH-TouchScreenFirmware #2051: [BUG] TFT35 V3.0 Calibration

ShiJuuRoku, 2021-07-08:
### Description

After performing the screen calibration, it seems that there is a problem with the Y axis in the lower left part of the screen

### Steps to reproduce

1. Update TFT Firmware to 0.27
2. Calibrate the screen
3. Touch the 'Menu' icon, at lower left part of the screen

**Expected behavior**
Go to menu

**Actual behavior**
When touching at the top of the icon > go to "Heat' panel
When touching at the middle of the icon > nothing
When touching at the  bottom of the icon, at the edge of the screen> go to the 'Menu' panel

### Hardware Variant
BTT TFT35 V3.0
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0.27.x.bin

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2021-07-09:
Try a new calibration using the stylus

ShiJuuRoku, 2021-07-09:
Trying multiple times, it's not better.

Again, the issue append only with the bottom left part of the screen.
The right part working perfectly.

It's weird, maybe the problem comes from the calculation of the coordinates ?

The calibration need only 3 points : top-left , top-right, bottom-right (then center to check consistency)

radek8, 2021-07-09:
Did it work well with the older version of FW?

ShiJuuRoku, 2021-07-09:
Not working with  0.25 version : BIGTREE_TFT35_V3.0.25.bin

radek8, 2021-07-09:
Then there is the HW problem with the touch layer. Return the display to the seller for a complaint.
The display FW is OK.

radek8, 2021-07-09:
Have you tried the reset file.txt?

ShiJuuRoku, 2021-07-09:
Yes i tried with the reset.txt file, and with the calibration option into the menu.

The screen worked with the stock FW when i bought it.

radek8, 2021-07-09:
Did you use the correct HW FW versions?

ShiJuuRoku, 2021-07-09:
Yes, I was taking this file : BIGTREE_TFT35_V3.0.27.x.bin

radek8, 2021-07-09:
And what's your display?

ShiJuuRoku, 2021-07-09:
[image]

oldman4U, 2021-07-12:
This is the correct firmware for this TFT. Have you been able to solve the problem? What was the original firmware version and have you tried that version?

ShiJuuRoku, 2021-07-13:
Ok, maybe it an HW issue ..
I don't know what is the original firmware, a bought the screen 2 years ago.

oldman4U, 2021-07-13:
Looks like a hardware issue. Retry what is written in the Readme with external power. If this does not help the hardware is likely broken.

closed 2021-07-14 by ShiJuuRoku