bigtreetech/BIGTREETECH-TouchScreenFirmware #2052: TFT mode no printing

AntoszHUN, 2021-07-10:
Hi! I downloaded the TFT firmware from GitHub 2-3 days ago. (Not the official release!) Since I am developing I am not using the official release. This is the current firmware on GitHubon. The machine does not print in TFT mode. Prints only in Marlin mode. What's the mistake? This is what he does in the video. It heats up the devices and does nothing afterwards. What the hell?

Made by: AntoszHUN

https://drive.google.com/file/d/1SoecoXfnzFSS8HPD6oPfoza6GwVbZv-K/view?usp=sharing

kisslorand, 2021-07-10:
Your printer doesn't start to print any gCode you try to print?

oldman4U, 2021-07-10:
Maybe related to #2040. AntoszHU probably uses his smart filament sensor also.

AntoszHUN, 2021-07-10:
> 
> 
> Your printer doesn't start to print any gCode you try to print?

No.

AntoszHUN, 2021-07-10:
> 
> 
> Maybe related to #2040. AntoszHU probably uses his smart filament sensor also.

I don't have a problem with freezing. Plus, I have a TFT70. And that works flawlessly. I simply described what was wrong. Video also supports it.

oldman4U, 2021-07-10:
Try to find the commit which broke it.

kisslorand, 2021-07-10:
I find it strange that it homes before heating. Was it like that when it worked?

AntoszHUN, 2021-07-10:
> I find it strange that it homes before heating. Was it like that when it worked?

This is always the case. It takes up the home position, heats up the devices, and then auto-leveling.

AntoszHUN, 2021-07-10:
> 
> 
> Try to find the commit which broke it.

Okay. But does it work for someone else or just me? I downloaded it from here. The developer version.

kisslorand, 2021-07-10:
The developer version was last updated in 12th of April 2021. You should use the master version.

AntoszHUN, 2021-07-10:
> 
> 
> The developer version was last updated in 12th of April 2021. You should use the master version.
> On your video it doesn't seem to be the developer version, I don't think at that time the printing progress bar with the stripes was implemented.

And this is the developer version. This is updated daily. Last merged 3-6 days ago.

kisslorand, 2021-07-10:
My internet must be broken...
[image]

AntoszHUN, 2021-07-10:
> 
> 
> My internet must be broken...
> [image]
> 
> Anyway, did you try the branch "master"?

Sorry! I wrote it wrong. I'm using the Master version. I always use it. Not the develop version.

radek8, 2021-07-11:
Will you send the file you are printing?

AntoszHUN, 2021-07-12:
> Will you send the file you are printing?

I do not understand. In Marlin mode, the machine prints correctly. There is nothing wrong with gcode files. Here the TFT firmware is faulty somewhere.

radek8, 2021-07-12:
You're the only one with a problem. The problem may be in your press file.

AntoszHUN, 2021-07-12:
> You're the only one with a problem. The problem may be in your press file.

Then I will download the Master version again and try again. But I can only see it in the evening. If the error disappears, I close the bug report.

radek8, 2021-07-12:
There may be an error in the FW display that responds to the code you have in the file. Is it so hard to send a sample file in which cyba appear?

AntoszHUN, 2021-07-12:
> There may be an error in the FW display that responds to the code you have in the file. Is it so hard to send a sample file in which cyba appear?

I'm not at home. Workers. I will send. What file do I need? What is the name?

radek8, 2021-07-12:
> > There may be an error in the FW display that responds to the code you have in the file. Is it so hard to send a sample file in which cyba appear?
> 
> I'm not at home. Workers. I will send. What file do I need? What is the name?

Will you send the file you are printing?

AntoszHUN, 2021-07-12:
> 
> 
> > > There may be an error in the FW display that responds to the code you have in the file. Is it so hard to send a sample file in which cyba appear?
> > 
> > 
> > I'm not at home. Workers. I will send. What file do I need? What is the name?
> 
> Submit the file you have trouble printing.

But it doesn't matter what file. You do not want to print anything from TFT mode.
[Utolso.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6803631/Utolso.zip)

AntoszHUN, 2021-07-12:
> But it doesn't matter what file. You do not want to print anything from TFT mode.
> [Utolso.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6803631/Utolso.zip)

There's a link to the video where you can see what the machine isn't doing.

radek8, 2021-07-12:
You can try commenting a line
M355 S1 P255
And try printing again?

[image]

AntoszHUN, 2021-07-12:
> You can try commenting a line
> M355 S1 P255
> And try printing again?
> 
> [image]

What's wrong with the M355 command? It has worked so far. No more support for firmware?

radek8, 2021-07-12:
I don't know, but your printer will stop at this command. Therefore, I suggest trying this to confirm or rule out this possibility.

AntoszHUN, 2021-07-12:
> I don't know, but your printer will stop at this command. Therefore, I suggest trying this to confirm or rule out this possibility.

All right. I will try without M355.

AntoszHUN, 2021-07-13:
> 
> 
> I don't know, but your printer will stop at this command. Therefore, I suggest trying this to confirm or rule out this possibility.

Hi! You were right Master! :) I didn't like the TFT display firmware in the M355 S1 P255 command in gxode. But why? Let's say it's normal to turn on the LED light when I turn on the machine? In TFT mode, the NeoPixel status bar does not work. :(

radek8, 2021-07-13:
Why are you closing the problem?
If there is a problem with the M355 command, you should leave the problem open so that it can be resolved.

AntoszHUN, 2021-07-13:
> Why are you closing the problem?
> If there is a problem with the M355 command, you should leave the problem open so that it can be resolved.
> Or, open a new problem on a broken M355 command

Therefore, I will not open a new issue. But I will comment here that the M355 command does not work with the TFT firmware. I will also change the title.

oldman4U, 2021-07-14:
I have no case light so I can not test it, but have you tried to send the same command via Terminal. There is some case light related code in the firmware and I am wondering if there is. general problem with it or "only" one in case it is sent as part of the print gcode.

AntoszHUN, 2021-07-14:
> I have no case light so I can not test it, but have you tried to send the same command via Terminal. There is some case light related code in the firmware and I am wondering if there is. general problem with it or "only" one in case it is sent as part of the print gcode.

I test after work in the evening.

kisslorand, 2021-07-14:
@AntoszHUN 
Do you have the same problem if printing from the motherboard SD?

AntoszHUN, 2021-07-14:
> @AntoszHUN
> Do you have the same problem if printing from the motherboard SD?

I do not know. I didn't try. I only print from a TFT SD card.

kisslorand, 2021-07-14:
If you can test it, please let us know of the results.

AntoszHUN, 2021-07-14:
> If you can test it, please let us know of the results.

I know. I will also test in the evening. I'm working now. I will also try the M355 command on a console.

digant73, 2021-07-14:
@AntoszHUN if you use SFS try to disable it or also try if PR #2061 solves your problem

AntoszHUN, 2021-07-14:
> @AntoszHUN if you use SFS try to disable it or also try if PR #2061 solves your problem

I use SFS. But that's not the problem. If I remove the M355 command from Gcode, it works.

digant73, 2021-07-14:
I would try to disable it. Also I would try also my PR.
In my printer your file is printed.

AntoszHUN, 2021-07-14:
Something in the TFT firmware is not working properly. I'll send the video soon. Is it normal for NeoPix to light up when I turn on the machine?

AntoszHUN, 2021-07-14:
Hi, I made a video. That's not how it used to work.

https://drive.google.com/file/d/1DkU3sXc85q8bamzdb9PkzAgTtQAtge9N/view?usp=sharing
https://drive.google.com/file/d/1DmMun3aajVtpkiQ2hMiGXGBjtbThSczp/view?usp=sharing

kisslorand, 2021-07-14:
Check PR #2064.

AntoszHUN, 2021-07-15:
> Check PR #2064.
> 
> @AntoszHUN
> Thanks for your perseverance, your videos helped finding the bug.

All right. I'm glad. I’m happy to be able to help with the work for 2 years now and can always add a little to make it a little better firmware. :) I see the fix. I'll try it in the evening. Now I go to work.

AntoszHUN, 2021-07-16:
The M355 bugfix, or #2064 request, fixes the bug. Thanks the @kisslorand !

PR #2064 (2021-07-19): M355 bugfix
### Requirements

BTT or MKS TFT

### Description

This PR fixes the M355 bug that was introduced by PR #2022, improves case light handling.

### Related Issues

Fixes #2052
M355 was not sent again if the first try failed.
Case light adjustment from outside triggered resending the settings again from TFT.
closed 2021-07-16 by AntoszHUN