bigtreetech/BIGTREETECH-TouchScreenFirmware #2053: TFT 24 V1.1 In Touchscreen Mode not interactable?

kazuimis, 2021-07-11:
Good day to all, I have recently added BTT TFT24 V1.1 to my Artillery Sidewinder X1 as I would like to have the dual features that BTT TFT has. I have sucessfully updated the firmware and able to do adjustments to get into Marlin mode. However when I want to use the Touch Display mode, for some odd reason I can interact with the UI.

Anyone could share with me is there a portion I should check to allow the interaction with the UI? I thought maybe my TFT touch display may have issue, but I allowed the interface to idle where the screen turns dim and when I interacted it brightens back. What is happing?

[Firmware used.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6796824/Firmware.used.zip)

radek8, 2021-07-11:
Send photos of the involvement. Send the Marlin configuration.

oldman4U, 2021-07-11:
Read the Readme and there especially the trouble shoot section.

kazuimis, 2021-07-12:
[image]

[image]
[image]

[Marlin Configuration used.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6798594/Marlin.Configuration.used.zip)

I went to look at the Readme section as well but not too sure if I have missed an info or not, @oldman4U mind if you could share which poertion you are reffering to?

@radek8 I have uploaded som images as well as the Marlin config I'm using!

radek8, 2021-07-12:
The display says Ready. So it's connected. What's your problem?

radek8, 2021-07-12:
Did you use the Reset file.txt?

oldman4U, 2021-07-12:
Like I wrote above. Troubleshooting.

But what is your problem? What happens when you press a button?

kazuimis, 2021-07-12:
@oldman4U When I pressed it, it does not respond to bring me to menu etc in touch screen mode.

@radek8 I just name an empty .txt file as Reset right?

oldman4U, 2021-07-12:
Do what is written in the ReadMe under Troubleshooting.

kazuimis, 2021-07-12:
Hi @oldman4U I have tried again on the Rese.txt and some how it worked 👍 not guessed like you mentioned keep trouble shoot! And thanks @radek8 I have fixed it.

closed 2021-07-12 by kazuimis