bigtreetech/BIGTREETECH-TouchScreenFirmware #2057: preheating of 2 nozzles

Pets27, 2021-07-12:
Hello,
I would like it to be possible to enter 2 nozzles and a hotbed (in the config file) in the preheat settings. I have a dual x axis and if I turn on the preheating then I have to set the second nozzle manually. I think more users would welcome this option.
Thank you Petr

oldman4U, 2021-07-13:
Maybe s custom gCode would be better in this case.

Pets27, 2021-07-13:
Yes, it would work. I only use two materials, Petg and ABS. If the second nozzle could also be set in preheating, I would set the material 1 ABS, 2 PETG, 3 PETG 2 nozzles, 4 ABS 2 nozzles in the config.
Then the preheating option would only be in one "window".
This would also be good for setting different temperatures when someone uses a material with a different temperature. For example ABS + PVA etc.

oldman4U, 2021-07-13:
You have plenty of custom gCodes you can use for features which are not available today.

Pets27, 2021-07-13:
OK. It was just a suggestion for improvement for printers with 2 nozzles.
For example:
preheat_name1: PETG PVA
preheat_temp1: TA220 TB210 B75
(TA - nozzle 1, TB - nozzle 2, B - bed)
To think about the future
Thank you Petr

oldman4U, 2021-07-13:
Maybe you are lucky and one of the contributors will add it one day. But I wouldn't bet on it so custom gCode will do the job definitively sooner.

gordo3di, 2021-07-14:
> Maybe you are lucky and one of the contributors will add it one day. But I wouldn't bet on it so custom gCode will do the job definitively sooner.

I tend to agree. We already have it added as a custom gcode since you have like 15 slots.

closed 2021-09-19 by stale[bot]