bigtreetech/BIGTREETECH-TouchScreenFirmware #2065: Temp Error on temp change

nor1bie, 2021-07-15:
### Discussed in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/discussions/2063

<div type='discussions-op-text'>

<sup>Originally posted by **nor1bie** July 14, 2021</sup>
Not sure if anyone else has seen this issue.
With the latest firmware (pre-built BIGTREE_TFT35_V3.0_E3.27.x  Jul 7. 2021 or somewhere around) for the TFT35 E3 V3, I see fatal errors when trying to print a Temp Tower with embedded temperature changes. Initially, I have a start temp of 205C set in gcode. After leveling, the gcode instructs to change to 220C, however it continues to print @ 205C (this maybe a gcode issue). After the 220C section (which is at 205C) encounter's the change to 215C a temp error (I believe timeout) is thrown. This is a fatal error and the printer halts. 
Using the same code feeding it through onboard SD slot, it seems to work without the fatal error. But, the temp change is still ignored and I have to change it manually throughout the print. 
My guess would be that on either side something with the temperature algorithm doesn't work correctly. 

The reason I start here is simply because it doesn't throw the error when printing through onboard sd.  

I have to further investigate the problem with not changing the temperature, but think there are too many cooks involved and mess up the dish. In the short time, since October 2020, I am using Marlin / TFT35, I have been scratching my head so much, that I became bald.
Many times I made changes on one end whilst the other end somehow managed to do something else. At least it did something not logical to me. It is very difficult to follow what device is doing what and how they interact, meaning not knowing what side did cause the issue, leading to mostly something that does only work half. Once I got to a point where things seemed to work in harmony, I upgraded the more better, more gooder firmware and the entire thing starts over again.
Most of the time you can't use the known working config file, someone decided to make the incompatible. 

Would be interesting to know if anyone else has a similar experience, since I could not find anything on Google pertaining to the temperature errors.</div>

oldman4U, 2021-07-15:
Maybe you want to close the discussion then and attach the gcode file here.

Improving things requires to changes also existing things sometimes. Best would be to stay with a firmware version you have no problems with.

nor1bie, 2021-07-15:
> Maybe you want to close the discussion then and attach the gcode file here.
> 
> Improving things requires to changes also existing things sometimes. Best would be to stay with a firmware version you have no problems with.

Hi, just did some more investigation. I came up with another perplexing example <g>. I have a plugin for cura that is supposed to decrement temperatures automatically and plug it into the Gcode at the appropriate places. Well, it doesn't plug anything! Yes, I enabled it. My guess, Cura 4.10 doesn't know how to handle the plugin, perhaps of it's age or? Anyway, the plugin doesn't work at all. I edited the Gcode manually and it does in fact change the requested temperature. 
So, one mystery is out of the way.
This is with the SD card to the Mainboard, haven't tested the TFT card slot yet.
Another good example of, you think something is making it easier / faster, but it just says, my name is "Murphy"! You really have to double check absolutely everything. 

I think I keep it open until I printed the object from the TFT slot and report.
I am sure there was fatal errors, since I tried at least 4 times.

nor1bie, 2021-07-15:
So, I went and started the same Gcode as with the onboard Gcode.
No error anymore.
The only thing that makes sense, is that the Cura plugin created some garbage, which I didn't bother to find and plugged it into the code.
I'll go ahead and close the issue.
Thanks

closed 2021-07-15 by nor1bie