bigtreetech/BIGTREETECH-TouchScreenFirmware #2069: TFT3.5 V3 on Rumba32 - no printer attached - Repetier v2

polymeer, 2021-07-20:
Hello friends,

I try to connect the BBT TFT3.5 v3 to a MKS Rumba32 v1.0.

Flashing firmware on both devices done. The display is attached as on the pictures shown. At the moment I have just the plain motherboard. Nothing is attached. No steppers, no endstops no additional power.

- Is the TFT working with Repetier?
- Is it a problem with the plain board?

[image]

[image]

discip, 2021-07-26:
Have you tried different baudrates on the tft in the settings under connection.
After the change you have to wait for a moment.

oldman4U, 2021-07-29:
@polymeer

@discip tries to help you. Please answer or close this ticket.

p-v-n, 2021-07-30:
I'm having the same issue with TFT35 V3 (with latest FW) on SKR E3 Turbo. Tried different baudrates in Marlin and in TFT settings but to no avail.
The only time when TFT stops displaying this message is when Octoprint connects to the board through USB. In that case TFT reports correct temperatures for hotend and bed, however is unable to issue any commands itself, e.g. changing temp, or homing. etc...
Currently I have these settings in Marlin:

```
#define SERIAL_PORT -1
#define BAUDRATE 250000
#define SERIAL_PORT_2 0
#define BAUDRATE_2 115200   // Enable to override BAUDRATE
```

oldman4U, 2021-08-01:
@ Msq001

Please close this ticket. The owner never replied.

Thank you