bigtreetech/BIGTREETECH-TouchScreenFirmware #2070: Error preventing compiling, `'PID_CMD_MARLIN' undeclared here`

lightmaster, 2021-07-21:
Getting an error that's preventing compiling. Referenced line was last modified as part of #1962 and commit 80f8963d344e6e68d6fcc47d6aa879641414865d

```
> Executing task in folder BIGTREETECH-TouchScreenFirmware: C:\Users\light\.platformio\penv\Scripts\platformio.exe run <

Processing BIGTREE_TFT43_V3_0 (board: STM32F207VC_0x8000; platform: ststm32; framework: cmsis)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/STM32F207VC_0x8000.html
PLATFORM: ST STM32 (14.0.1) > STM32F207VC (96k RAM. 256k - 32K Flash)
HARDWARE: STM32F207VCT6 120MHz, 96KB RAM, 224KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-cmsis 2.50501.200527 (5.5.1)
 - framework-cmsis-stm32f2 2.2.3
 - tool-ldscripts-ststm32 0.1.0
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Warning! Cannot find a linker script for the required board! An auto-generated script will be used to link firmware!
Warning! Cannot find the default startup file for stm32f207vct6. Ignore this warning if the startup code is part of your project.
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Generate language_xx.ini files:
Processing: language_am.h: Total keywords found:317, File generated:language_am.ini
Processing: language_ca.h: Total keywords found:317, File generated:language_ca.ini
Processing: language_cn.h: Total keywords found:317, File generated:language_cn.ini
Processing: language_cz.h: Total keywords found:317, File generated:language_cz.ini
Processing: language_de.h: Total keywords found:317, File generated:language_de.ini
Processing: language_du.h: Total keywords found:317, File generated:language_du.ini
Processing: language_en.h: Total keywords found:317, File generated:language_en.ini
Processing: language_es.h: Total keywords found:317, File generated:language_es.ini
Processing: language_fr.h: Total keywords found:317, File generated:language_fr.ini
Processing: language_gr.h: Total keywords found:317, File generated:language_gr.ini
Processing: language_hu.h: Total keywords found:317, File generated:language_hu.ini
Processing: language_it.h: Total keywords found:317, File generated:language_it.ini
Processing: language_jp.h: Total keywords found:317, File generated:language_jp.ini
Processing: language_pl.h: Total keywords found:317, File generated:language_pl.ini
Processing: language_pt.h: Total keywords found:317, File generated:language_pt.ini
Processing: language_ru.h: Total keywords found:317, File generated:language_ru.ini
Processing: language_sk.h: Total keywords found:317, File generated:language_sk.ini
Processing: language_sl.h: Total keywords found:317, File generated:language_sl.ini
Processing: language_tc.h: Total keywords found:317, File generated:language_tc.ini
Processing: language_tr.h: Total keywords found:317, File generated:language_tr.ini
Processing: language_uk.h: Total keywords found:317, File generated:language_uk.ini
Total language files processed: 21
Compiling .pio\build\BIGTREE_TFT43_V3_0\FrameworkCMSIS\system_stm32f2xx.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\Startup\stm32f2xx\startup_stm32f2xx.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\base64\base64.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\cmsis\stm32f2xx\system_stm32f2xx.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\misc.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_adc.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_can.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_crc.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_cryp.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_cryp_aes.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_cryp_des.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_cryp_tdes.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_dac.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_dbgmcu.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_dcmi.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_dma.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_exti.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_flash.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_fsmc.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_gpio.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_hash.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_hash_md5.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_hash_sha1.o
[... 116 lines cut ...]
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\MeshTuner.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\MeshValid.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\More.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\Move.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\NotificationMenu.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\ParameterSettings.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\PersistentInfo.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\Pid.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\Popup.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\PowerFailed.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\PreheatMenu.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\Print.o
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\PrintingMenu.o
TFT\src\User\Menu\Pid.c:22:36: error: 'PID_CMD_MARLIN' undeclared here (not in a function); did you mean 'ICON_MARLIN'?
 const char *const pidCmdMarlin[] = PID_CMD_MARLIN;
                                    ^~~~~~~~~~~~~~
                                    ICON_MARLIN
Compiling .pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\RRFMacros.o
*** [.pio\build\BIGTREE_TFT43_V3_0\src\src\User\Menu\Pid.o] Error 1
====================================================================== [FAILED] Took 33.76 seconds ======================================================================
Environment         Status    Duration
------------------  --------  ------------
BIGTREE_TFT43_V3_0  FAILED    00:00:33.764
================================================================= 1 failed, 0 succeeded in 00:00:33.764 =================================================================The terminal process "C:\Users\light\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

digant73, 2021-07-21:
PID_CMD_MARLIN is defined in Configuration.h. Verify it is present in your project first. Otherwise make a git refresh

lightmaster, 2021-07-22:
Apparently VSCode decided that was not an important line to update when I merged it into my fork. Changed `PID_CMD` to `PID_CMD_MARLIN` and it compiled fine.

closed 2021-07-22 by lightmaster