bigtreetech/BIGTREETECH-TouchScreenFirmware #2072: What could go wrong with the TFT70 display?

AntoszHUN, 2021-07-25:
I recently got a TFT70 display from you.

The displayed image is not good. Confused.
I have another TFT70 display. I also took a picture of it. The worse pictures were taken from the newer display.
What makes the display so confusing? It is visible in all areas. Gray, red, green, blue. The color doesn't matter. If I look a lot my eyes get confused too.
Could it be a software or hardware failure?

[image]
NEW_TFT70_03
[image]
OLD_TFT70_02
[image]
NEW_TFT70_02
[image]
OLD_TFT70_00
[image]
NEW_TFT70_00
[image]
NEW_TFT70_01
[image]
OLD_TFT70_01

Made by: AntoszHUN

oldman4U, 2021-07-25:
How can this be firmware related when you use the same firmware with different results on 2 pieces of hardware?

Try standard Icon theme to see if you get different results also.

AntoszHUN, 2021-07-25:
> 
> 
> How can this be firmware related when you use the same firmware with different results on 2 pieces of hardware?
> 
> Try standard Icon theme to see if you get different results also.

It doesn't matter which icon theme I use. The image on the display is so bad. I hope you can see the difference. The last two pictures are well reproduced. Is the error not software?

digant73, 2021-07-25:
Try to unplug the display flat cable, clean the contact part of the cable and the connector and plug the cable

AntoszHUN, 2021-07-25:
> Try to unplug the display flat cable, clean the contact part of the cable and the connector and plug the cable

I'm beyond that. I've tried everything. I also looked at the solderings. :(

oldman4U, 2021-07-25:
You have 2 displays. Replace the one which looks good with the faulty one and make sure both use the same firmware.

AntoszHUN, 2021-07-25:
> You have 2 displays. Replace the one which looks good with the faulty one and make sure both use the same firmware.

They both have the same firmware.

oldman4U, 2021-07-25:
Which brings me back to my very first question...

AntoszHUN, 2021-07-25:
> 
> 
> Which brings me back to my very first question...

I was just wondering if anyone else had encountered the error.

closed 2021-07-25 by AntoszHUN

radek8, 2021-07-25:
It seems that BTT has more component suppliers, including LCD panels, and the quality is different. It looks like a poor quality LCD panel.
It was also visible in the trouble they had with SKR V2. In the event of a MOSFET supply failure, they used other types and then it burned ...

AntoszHUN, 2021-07-25:
> 
> 
> It seems that BTT has more component suppliers, including LCD panels, and the quality is different. It looks like a poor quality LCD panel.
> It was also visible in the trouble they had with SKR V2. In the event of a MOSFET supply failure, they used other types and then it burned ...

Yes. And that's a big problem. Many will buy elsewhere because of such mistakes. I do not. Because I want to solve problems. The only problem here is the lack of detailed testing.