bigtreetech/BIGTREETECH-TouchScreenFirmware #2074: [FR] (feature request title)

discip, 2021-07-25:
**Is your feature request related to a problem? Please describe.**
Since most screens are relatively small, the file names can get very long.
I do not see any benefit of displaying the file type/extension.

**Describe the solution you'd like**
I think it would look even cleaner, if the extention **`.gcode`** was hidden.

file name before   |   |  file name after
-|-|-
*example.gcode*|**=>**|*example*

oldman4U, 2021-07-26:
Is there any disadvantage by showing the file extension?

discip, 2021-07-26:
What kind of disadvantage do you have in mind?
I do not know, if I am the only one, but I prefer it to have clean and concise UIs. 😁
Does anyone use files that do have extensions other than **`.gcode`**. 🤷‍♂️
If that is not the case, why should precious space be cluttered with useless *'information'*?
I hope, I made my point clear this time. 😅

kisslorand, 2021-08-06:
What if there are users who use different extensions? "NylonClip.1mm", "NylonClip.5mm", "NylonClip.12mm", etc
There are millions of users, you never know what filenames do they use.

discip, 2021-08-06:
@kisslorand
That's exactly what I asked about in the last post. 🤭

The least I want, is to box someone in.

I am sure there is a way to only remove the **`.gcode`** extension.
And furthermore one more option in the **`config.ini `** could make this feature optional. 😁

regards

Hukuma1, 2021-08-12:
+1 for the option to hide in `config.ini`. "Hide known extensions"

kisslorand, 2021-09-03:
What would those extensions be?

Hukuma1, 2021-09-03:
hide.extension="gcode,1mm,5mm,12mm"

Let users fill it in?

kisslorand, 2021-09-03:
I meant what would be those _**known**_ file extensions?

Hukuma1, 2021-09-03:
Just stick in `gcode` and the ability to add more manually please.

kisslorand, 2021-09-03:
I guess `gco` could be in that list as well.

discip, 2021-09-14:
@kisslorand
Good evening Sir,
are we allowed to expect a PR on this one any time soon? 😁
No offense.

kind regards

kisslorand, 2021-09-21:
I spent two hours on this and actually the work needed for this outweighs its benefit. If needed, I can explain it why but that will not change the fact that it is not as easy at it seems.
So I will leave this to other developer to implement it if he/she has some spare time to spend on this.

discip, 2021-09-21:
@kisslorand
Good evening Sir,
thank you very much having tried getting this done. 👍
Maybe someone else has a simpler (not as time consuming) solution for this.

kisslorand, 2021-09-21:
> Maybe someone else has a simpler (not as time consuming) solution for this.
 That is quite possible!

discip, 2021-12-11:
@digant73
## Thanks a ton! 👍🥇

kind regards

PR #2284 (2021-12-14): Configurable common file extension suppression + bugfixes
**IMPROVEMENTS:**
* Configurable suppression (hide) for common gcode file extension: new param "filename_extension" added in config.ini. If set to 0 (false), any file extension starting with ".g" or ".G" (e.g. ".g", ".gco", ".gcode" etc...) will be hidden in both List Mode / Icon Mode menu. That setting can also be changed from Screen->UI menu. It implements #2074

**BUGFIXES:**
* Fixed missing display of filenames in sub folders when navigating from onboard SD on latest Marlin fw. It fixes #2182
* Minor bug fixes and optimization

reso [... cut]
closed 2021-12-11 by discip