bigtreetech/BIGTREETECH-TouchScreenFirmware #2075: Ender 5 TFT 43 Invert Z How?

LTSpenny, 2021-07-26:
Hello, I have a TFT43 installed onto an Ender 5 Plus, I'm trying to set it up so that while using the touch screen when I press the Z up the bed will move up and z down arrow will move z down. ATM it does the opposite, I've edited the config.ini to invert the movement with both the Y and the Z so that they move to match the arrows on the tft but only the Y is working. I've also tried inverting the drivers on the marlin firmware which gets the result I want but printer won't home up to meet BL Touch probe, just try's going down. any info would be appreciated, I've included a link to my config files for both marlin and tft. Thank you!

**specify the device variant related to your question.**
Ender 5 Plus on SKR2 with BL Touch and TFT 43 Running Latest Marlin

[Config Files
](https://drive.google.com/file/d/1rQKb9UAxv28O4ijbHZzLpTzDDkPqjhER/view?usp=sharing)

radek8, 2021-07-26:
The current logic is that you move the nozzle down to the bed or up from the bed.
Ender 5 does not move the nozzle but the bed.
Could it be enough to swap the names on the Z Up and Z Down icons?

radek8, 2021-07-26:
The movement of the bed on the Z axis on the Ender5 has been handled in the past.
But I don't know how it turned out.
Try looking in closed tickets.

LTSpenny, 2021-07-26:
Will do thank you.

LTSpenny, 2021-07-26:
Been looking through the issues and all i can find is possibly swapping the icons? Would this be advisable? Searched through z logic, ender 5, invert z and thats about all i could find or at least seemed to relate to issue im trying to fix. Maybe im not using the right terminology?

radek8, 2021-07-26:
It doesn't matter if the nozzle moves up or the bed down.
The principle is still the same. The gap between the nozzle and the bed increases or decreases.
The only difference is in the shape of the icon.
Or am I wrong? Is there another problem?

LTSpenny, 2021-07-26:
I think so? The up icon has a z+ which in the ender 5’s case is moving the build plate down. And the same reversed for the z- 

im good with just swapping the pictures just wanted to make sure it was as simple as that.

discip, 2021-07-26:
@LTSpenny
So, ... this is solved?
In that case please close. 😊
Thanks
regards

closed 2021-07-26 by LTSpenny