bigtreetech/BIGTREETECH-TouchScreenFirmware #2076: [Question] config file illigal flash, not updating config

LTSpenny, 2021-07-26:
TFT 43 On Ender 5 Plus. Can only update TFT if i have the firmware.bin, config file in root and a reset.txt. If i try to update only config I receive the illegal flash message and have to re update with all files. If i dont include the reset.txt eother update fails or screen becomes unresponsive. Ive updated successfully with all files and reset.txt but im tinkering with config trying to figure iut some invert direction issues and getting baudrate default set but those changes wont stick and its a bit of a pain to re upload all files re upload all files to card every time instead of just updating the config. Any thoughts appreciated

[Config files](https://drive.google.com/drive/folders/1lCN9-IQN0YD23wVPorg92Jh2E2i6YiXe) config files

oldman4U, 2021-07-26:
Are you formatting the SD Card you use for flashing the firmware on macOS?

LTSpenny, 2021-07-26:
Reformatting the card every single time on mac os. I had the format card issue on my tft35 so expected that just not the whole having reset.txt and full set of update files. Tried multiple cards too.

oldman4U, 2021-07-26:
I have the same problem also using macOS. Seems we have to go the long way and update everything all the time which works for me. Trying to use a previously used card or just a part does not work for me but for all other users using Windows.

oldman4U, 2021-07-26:
Changing settings in config.ini should make a difference.

LTSpenny, 2021-07-26:
Right but changing settings on config.ini is not actually making changes. Any suggestions on what else i could do to troubleshoot that? So far like i said the two that i noticed made no change even though configured in .ini are the baud-rate and the inverting axis for touch control screen. Thanks again!

radek8, 2021-07-26:
Can you create a video on which the configuration using config.ini will be captured?

Can you format the SD card and copy the Config ini file to it on a Windows PC and verify the difference in behavior against the SD card formatted with the MAC OS?

LTSpenny, 2021-07-26:
Aye should be able to do that. I can just format the card on each OS and copy files over right. No need to “create” or edit the config on each OS prior to the above directions? Will do tonight.

radek8, 2021-07-26:
This can also be affected by the software in which you edit the file.
Some programs may insert additional tags into the file that may not even be visible ...

A suitable file for editing is Notepad ++
https://notepad-plus-plus.org/

LTSpenny, 2021-07-27:
That seems to have been the issue. It also solved my inverting z issues on screen. For anyone dealing with this. Try using notepad ++ to edit your software and chuck one more on the mac os giving problems when formatting cards. Thanks all.

closed 2021-07-27 by LTSpenny

radek8, 2021-07-27:
So where was the problem?
In the formatting that MacOS does, or in the program you edited config.ini?

LTSpenny, 2021-07-27:
Seems the issue was with the formatting on mac.  I saved the files and put them on my google drive. Downloaded on windows. Formatted card. Saved files and everything worked as it should. Even started saving the settings on the config. File. I dont quite understand it but at least i know how to more efficiently do the updates now.

oldman4U, 2021-07-27:
Which software have you been using on macOS to edit config.ini, TextEdit?

LTSpenny, 2021-07-27:
@oldman4U a mix of visual code and Meld.

radek8, 2021-07-27:
Unfortunately, I don't have anyone around who has a Mac to see what's different on the card

oldman4U, 2021-07-28:
I use BBEdit or TextWrangler and at least do not see the issue with the config.ini.