bigtreetech/BIGTREETECH-TouchScreenFirmware #2077: [BUG] (short description)level corners fails after new update

BobCraig10, 2021-07-26:
### Description
I just updated the tft35v3 firmware, after updating, and trying to level the corners in the ABL menu, the BL Touch fails on the 3rd probe. it says probe failed. When the bed id being probed the pin doesnt stow with the new firmware. On the 3rd corner probe, the BL touch stays red, even after the probe touches the bed. 

### Steps to reproduce

1. [First Step]go to abl menu
2. [Second Step] click on L Corner
3. [and so on...] Click start

**Expected behavior**
I expect it to successfully probe 4 corners

**Actual behavior**
when it gets to the right rear corner, the bl touch stays red when the probe touches the bed, and gives an error on screen "probe failed"

### Hardware Variant
tft35 v3
### TFT Firmware Version & Main Board Firmware details BIGTREE_TFT35_V3.0.27.x.bin    Marlin 2.0.9 bugfix

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

I went back to the previous firmware for the tft35 v3, and the issue is gone

oldman4U, 2021-07-26:
Previous and new means exactly what?

BobCraig10, 2021-07-26:
> Previous and new means exactly what?

The firmware for the tft35v3 is showing here that it was updated 3 days ago.  Which is new to me.   Previous is the one before that which I had saved on my harddrive. from a month ago.  bin, config, and language/icon pack

oldman4U, 2021-07-26:
It would be important to know the exact PR which broke it.

radek8, 2021-07-26:
I have the last compilation, and this feature works without errors.

radek8, 2021-07-26:
Can you send a video?

digant73, 2021-07-26:
the bug was introduced by some changes in LevelCorner in #2047. Those changes were also the origin of Radek's PR #2066.
The bug is fixed in PR #2073 (the old code not affected by the bug is restored)

BobCraig10, 2021-07-26:
> Can you send a video?

I can reload the version with the issue back on my machine if you need later tonight. I have some prints going right now that I can't disturb, but I could do it in a couple hours if needed.

BobCraig10, 2021-07-26:
> the bug was introduced by some changes in LevelCorner in #2047. Those changes were also the origin of Radek's PR #2066.
> The bug is fixed in PR #2073 (the old code not affected by the bug is restored)

This means you found the issue and I don't need to re load that version to get a video? Thanks.

BobCraig10, 2021-07-26:
Thank you everyone. It was great to see how the issue was jumped on right away. I'm sorry I wasn't more clear about which versions I was using.   I only saw the dates, and the main firmware numbers, not the bugfix? numbers.

closed 2021-07-26 by BobCraig10