bigtreetech/BIGTREETECH-TouchScreenFirmware #2078: [Q] What could cause vertical noise after update?

brunoosti, 2021-07-27:
What on this firmware (and branches) could cause vertical random noise on the TFT screen? (most noticeable on gray buttons) 
Like refresh rate or something like it?

I have a sidewinder X1 and upgraded the TFT from as many sources I could find, and all leads to this vertical noise on the screen. Stock TFT firmware shows no interference at all. 

Already revised the wires connections inside the printer and disconnected fans with no change to the interference.

discip, 2021-07-27:
Did you compile it yourself?
Nothing like that here (same machine)! 🤷‍♂️
Try the following one:
[mkstft28.bin.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6884925/mkstft28.bin.txt)
***!rename first!***

brunoosti, 2021-07-27:
The first I got was from digant on thingiverse (excelente work, by the way). Than I tested others, like the one with a dinosaur, and a couple more. Finally I installed a different UI from 3d nexus which didn't show any interference. One thing I noticed in common between the 3d nexus and the stock firmware is that both have the buttons as .bin files while the branches from this are .bmp. Would thar even make any difference?
It is like some electromagnetic interference (vertical thin lines dancing along the screen) and the screen seams a bit more "pale", with not so much contrast as it could have.

discip, 2021-07-27:
Have you tried the above attached?

brunoosti, 2021-07-27:
Just did. Almost no noise, but much better than before. What kind of sorcery is this?

discip, 2021-07-27:
Have you changed the mainboard?
What else is connected to the TFT?
On mine I removed the filament runout sensor.

brunoosti, 2021-07-27:
Everything is sock. 
Also I tested disconnecting the sensor plug. And the board fan. 
Had to disconnect the reset cable from the TFT to be able to flash new Marlin fw (though digant's tft fw says it is not required to open the printer). But the noise was already there before I did it.

brunoosti, 2021-08-09:
Could it have something about grounding the printer? (though it doesn't how in other TFT fw)

discip, 2021-08-10:
Could you please post a picture of the backside of the TFT when configured with BTT's firmware?

brunoosti, 2021-08-10:
Sure : ) Thank you for your interest in this issue!
[image]
[image]

brunoosti, 2021-08-18:
Video with a lot of interference on: https://photos.app.goo.gl/6DeLjDvYfgxnGG8v9

discip, 2021-08-18:
Good afternoon @brunoosti,
have you tried removing the filament run out sensor?
In the 1st picture the upper left.

brunoosti, 2021-08-18:
yep. Removed every single wire possible.

discip, 2021-08-18:
@brunoosti
Have you tried unscrewing the TFT and the PCB with USB and TF card from the main body?
Make sure, that it does not touch the metal body while powering on.

brunoosti, 2021-08-18:
Will do that! Read somewhere, about someone else's other issue, that a plastic washer should be placed between the board and the frame (I think it was about the touching not being properly registered)

brunoosti, 2021-08-18:
No difference, as expected. If there is no noise with other TFT fw, the difference is in this one. Or my board (v 1.2). I looks so much as it's some brightness issue. The fw that don't have this setting in the config file also present no noise. This one have it, but the screen can't interpret it.

oldman4U, 2021-08-22:
Have you tried to remove ALL cables and connect only 5V and GND like described in the ReadMe?

Have you tried an older fw from this repository?

brunoosti, 2021-08-22:
Couldn't find what you refer to in any of the 4 files in the folter readme. Either way, I had the printer disconnected from the wall plug and only USB though the main board and it had the noise.
I've tried other instances of the fw, all showing this strange interference. 

NO interference: 
https://github.com/artillery3d/sidewinder-x1-tft-firmware  (stock)
https://3d-nexus.com/forums/resources/artillery-swx1-marlin-1-1-9-advanced-firmware-and-gui.2/
https://www.thingiverse.com/thing:3856144

With interference: 
https://www.thingiverse.com/thing:4294049
https://github.com/wgcv/RAWR-TFT-Firmware-Artillery3D
https://3dprintbeginner.com/custom-screen-firmware-for-sidewinder-x1-and-genius/

What differentiate those??

oldman4U, 2021-08-23:
This ReadMe.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

brunoosti, 2021-08-23:
Oh ok! 
As I don't have the tools to make the proper connector, did something "similar" just now: reinstalled the TFT fw with only the USB connection to the printer (in another words, though the main board). Same interference. 
Imagine having to open the printer every time to flash the TFT fw, though.

oldman4U, 2021-08-23:
This does not help. You have to provide power from another source than the mainboard.

brunoosti, 2021-08-23:
Sorry... I understand.. 
for now this is a dead end for me, as I'd risk frying the TFT without the proper connection.

oldman4U, 2021-08-24:
So you need someone else with the same problem doing this test. How many other users do you know with the same problem who could do this test?

brunoosti, 2021-08-24:
Not many. A couple follows/likes in one or another posts on thingiverse/facebook/reddit.

oldman4U, 2021-08-24:
Given the number of sidewinder sold, there must be hundreds with this issue. Finally all which are upgrading the firmware of the TFT. One has to be willing to do all the tests needed to find the problem or it will not be solved.

brunoosti, 2021-08-24:
Maybe It doesn't bother people that much (or they simply don't know where to look for a solution). 
Right now, at this moment, I'm printing something and the noise is almost unnoticeable (it happens). Go figure... Maybe It is something with my electric grid (the strange thing being it does not affect some other firmware). 
It's a fact a couldn't find anything similar anywhere online.

oldman4U, 2021-08-24:
Software changes over time. Most of the time it improves things. Sometimes a change causes issues but your issue is more likely a hardware issue or a combination of several things.

BTT turned off all serial ports by default which solved issues like interrupted or failed prints in case of a problem with EMI. If a user with such an issue now activates one or several serial ports he will correctly claim that activating the serial port causes issues. But he is wrong. Very likely your problem is something similar and a problem which always existed is "visible" when using the firmware from this repository.

closed 2021-10-31 by stale[bot]