bigtreetech/BIGTREETECH-TouchScreenFirmware #2085: [Q] (Is there a way to show the chamber temp on tft50 in touch mode. Using the latest version of firmware

Tezzatron81, 2021-07-31:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

radek8, 2021-07-31:
actual temperature and target temperature of CHAMBER will be displayed on the top title bar.
It has once been a demand to change the display with the temperature of the bed. It has not been implemented yet.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1182

Tezzatron81, 2021-07-31:
Do u mean while its printing it shows?

radek8, 2021-07-31:
no

oldman4U, 2021-08-02:
PR #2084

Please test and let us know.

PR #2084 (2021-08-06): Clean up and optimize printing screen
- Clean up and reduce RAM usage in the Printing screen.
- Optimize the Printing screen to load faster.
- Prevent unnecessary status icon redraw in the Printing screen.
- Status icons on the Printing screen are now clickable after the print is finished.
- Clicking on the Layer/Z-layer status icon now opens the Babystep menu.
-  Misc. clean-up and reduce global variables.
-  Add chamber temperature status in status screen and printing screen (#1182). The bed heating icon will toggle between bed and chamber status If chamber heating is available.

Resolves #1 [... cut]
closed 2021-08-06 by bigtreetech