bigtreetech/BIGTREETECH-TouchScreenFirmware #2086: [BUG] v27 Flashing Screen SKR2

somejavadev, 2021-07-31:
### Description

On the SKR2 rev B board connecting the TFT35 E3 version works fine on the v26 release https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.26 but upgrading to the latest master or v27 release causes the screen to flash black and white after a successful upgrade. After adding the `reset.txt` file the printer boots up and shows the calibration screen, after completing the calibration the screen goes back to flashing, this time the controller/dial light is on which wasn't on before. Having a look at the SD card the files have all been renamed to `.CUR`

### Steps to reproduce

1. Connect the TFT35 E3 to the SKR2 (EXP1,EXP2 and TFT)
2. Add the the content of the attached zip (folder: sd-card-content-for-upgrade) to a SD card and perform the upgrade of the TFT
3. Wait for the upgrade to be complete.
4. The screen will start flashing. The flashing is almost like the backlight turning on and off constantly.
5. Turn off the printer and add a `reset.txt` file to the SD card
6. Turn on the printer and notice the calibration screen without any flickering.
7. Complete the calibration and notice the flicker/flashing starts again. 

**Expected behavior**

There shouldn't be any flickering and the display should be usable.

**Actual behavior**

The display flickers/flashes and is unusable.

### Hardware Variant

TFT35 E3

### TFT Firmware Version & Main Board Firmware details

Working version: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.26
Not working version: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.27 or current master branch

Mainboard: SKR2 Rev B

Marlin 2.0.9.0

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

I have both EXP1 and EXP2 cables as well as the TFT cable connected as per SKR2 documentation.    
I am trying the Unified Theme.    

[debug.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6911769/debug.zip)

oldman4U, 2021-07-31:
Disconnect the currently connected cables and connect external power, like described in the Troubleshooting section of the ReadMe: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#troubleshooting

somejavadev, 2021-08-01:
@oldman4U I can fix this by uploading the older firmware to the SD Card and "upgrading" it back to v26, how do I get the latest firmware to work, the v26 on the releases page is nearly 1 year old and there has been a number of improvements since then.

somejavadev, 2021-08-01:
@oldman4U closing this issue the EXP1 and EXP2 cables were swapped around causing this. Only strange thing is the display was working fine on v26.

closed 2021-08-01 by somejavadev