bigtreetech/BIGTREETECH-TouchScreenFirmware #2087: Screen freezes when plugging an certain usb drive.

xXDexorioXx, 2021-08-01:
Just installed an BTT SKR MINI E3 V2.0 with an BTT TFT35 E3 V3.0 on an Ender 3 Pro.

When plugging in an SanDisk Ultra USB 3.0 64GB with an gcode file it shows that it reconised the udisc but freezes in that moment.

If i but in an "lower cost" USB stick in with the same capacity it acts normal.
Any idea what it could be? To much power draw because of the higher read/write speed of the sandisk?

radek8, 2021-08-01:
How FS are the disks formatted? Do they both have the same FAT32 format?

oldman4U, 2021-08-03:
@xXDexorioXx

Radek8 tries to help. Please answer or close this ticket.

oldman4U, 2021-08-05:
@Msq001

User does not respond to his own ticket. May you want to close it. Thank you

closed 2021-08-05 by xXDexorioXx