bigtreetech/BIGTREETECH-TouchScreenFirmware #2089: Can’t seem to get L Corner Menu working

LTSpenny, 2021-08-02:
In TFT43 connected to SKR 2 with a BL Touch Probe, I have a menu called L Corner under ABL menu (video). On my TFT35 this works. Trams to corner. Probes with BLTouch and gives value. On TFT43 it just homes and does nothing. I’ve compared files with my ender 3 configs to my ender 5plus’s and cant seem to find the setting. Any assistance appreciated. Ive attached a video of the issue as well as what my ender 3 does. Also included all my config files for both machines. 

What my ender 5 does. 
https://youtu.be/cpoPQIBK3U4

My ender 3
https://youtu.be/BA_pAyQEJSs

Config Files. 
https://drive.google.com/drive/folders/1lCN9-IQN0YD23wVPorg92Jh2E2i6YiXe

LTSpenny, 2021-08-02:
Well tinkering with it i changed the distance to 50 and that seems to have it working. Is there a place i can edit this in the firmware? Im a big fan of resetting the prom chip after maintenance and one less thing id have to reset would be nice. Is it just in the l corners section?
#Level_corners_inset_lfrb {30, 30, 30, 30}?

radek8, 2021-08-02:
The distance from the edge of the bed should not be less than the most important offset value of the probe

closed 2021-08-02 by LTSpenny