bigtreetech/BIGTREETECH-TouchScreenFirmware #2090: [BUG] TFT35 GD firmware update error

poumsoon, 2021-08-02:
There was a bug in the tft35 GD v2 version firmware, so I tried to update it.
However, it does not proceed in the "Deleting old ROOT_DIR...." step.
[image]

radek8, 2021-08-02:
How is the GD version different from standard vefze 2.0?
You may need to upload a new bootloader.

bigtreetech, 2021-08-04:
Hello, Please try to create a file named “reset.txt” in the SD card root directory to reset all the parameter settings.

oldman4U, 2021-08-04:
@bigtreetech I am sure a user reads the ReadMe before he opens a ticket here...;-) What is the difference to the standard TFT35 v2?

poumsoon, 2021-08-05:
Yes, the firmware update was successful.
thanks for the quick fix
In the future, we hope that new product buyers will not become bug testers.

closed 2021-08-05 by poumsoon

radek8, 2021-08-05:
> Yes, the firmware update was successful.
> thanks for the quick fix
> In the future, we hope that new product buyers will not become bug testers.

What bug fix do you mean?