bigtreetech/BIGTREETECH-TouchScreenFirmware #2101: [Q] Softwarelimit of heated Chamber max Temperature

MarkusThur, 2021-08-05:
**Your question or details you need. Please describe.**
I am looking into the "limiting" of heated chamber max Temperature. For my project I need to be able to set heated chamber to aprox 200 to 220 degree celsius. So any idea where to find that?

**Additional context**
I am aware that for security reasons, even if a higher chamber temperature than 100°C is specified in .ini file, the max Temp gets limited to 100°C. For my project i want to intentionally override this limit / increase it to 250°C.
So in a first aproach I just need the position in the code, where to change this. (I will create a branch which) provides that.
In a seconnd step I want to think about how to integrate that safely into the firmware, like having a aditional "flag" HTchamber and a compiler warning while compiling with "HTChamber" Flag set

oldman4U, 2021-08-05:
Are you going to open a grill and smoke bar with vertorek #2049?

;-)

Happy BBQ

oldman4U, 2021-08-05:
Have you seen this and does it help?

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/discussions/1567

MarkusThur, 2021-08-05:
@oldman4U 
Thank you, will have a look there and hope to find it for the chamber there also.
And tbh idk why he wants to heat the bed that high, but chamber and bed to aprox 200°C you need for this materials e.g.:
https://www.igus.de/product/17742?artNr=J350-PF-0175-0250

Same counts for materials like peek, It's not only the hotend you need to get up, you also need to heat up the bed, to get a proper bed adhesion and you need to get the chamber up, iot prevent wrapping. (At least this is the esiest way).

As a thumb rule, you will the followinf for nearly all thermopasticts:
<hotend Temp> = <max perm Temp> + [120 - 200] K
<hotbed Temp> = <max perm Temp> - [5 - 25] K
<chamber Temp> <= <hotbed Temp> && <hotend Temp> - <chamberTemp> < 200K

It's just a Thumb, but works. examples:
PLA max perm temp 65°C => hotend = 180 - 240°C; bed = 40 - 60°C; Chamber= Ambient temperature is just right, but somewhere above 24°C gives better results
ABS max perm Temp 120°C => hotend = 235 - 270°C; bed = 90 - 110°C; Chamber = 35 - 70°C
IGUS J350 max perm Temp 180°C => hotend = 350 - 360°C; bed = 155 - 175°C; Chamber = 160 - bedTemp°C
And so lets guess for peek:
(Verotorek aready explained it properly. Especially peak is a optimized semicrystaline polymere, which helps us a bit. and allows lower temperatures / more easy printing.  The melting point and the glass transition poit are pretty far away from each other. It's enough to have bed and chamber above glass transition point and can be up to permanent stable temperature. It's a high perfomance polymere, like Polyamid hotmelt also is.. but lets try the guessing with numbers from above)
Stable temp of peek is 260°C (and we keep in mind glass transition already is at 155, where the state changes to have more amorphous than crystaline components in the semicrystaline compound).
hotend = 380°C - 460°C
bed = 235 - 255 °C
because of the special characteristics you can go down to 155°C, but bed adhesion will increase the higher the bed will be tempered
Chamber = just below or at bed temperature
(A Chamber temp higher than bed temp is nonsense as the bed will heat then to chamber temp anyways, so you'd not need a heated bed :-))
This printer design proposed by a emloyee of IGUS for example uses exactly that trick, only heating the chamber and having bed = chamber temp... That design you can have a heated chamber and define it as heated bed for the controller:
https://blog.igus.de/do-it-yourself-3d-hochtemperatur-drucker-selber-bauen/

A misconception I regularly find is about wrapping, there are different reasons for it, but mainly it occures if <shrink force> > <bed adhesion force>.   The shrink force is greater the higher the temperature difference between hotend and ambient is, unless you make it to shock cool the material below glass transition, where it just can't deform. But then you have issues with layer adhesion and/or internal forces in the part.
Higher bed temperature increases activation level of the surface and holds the thermoplast open longer. (In general a longer "open" time invreases adhesion, this counts for nearly every adhesi). There maybe "sweetspots" for specific materials / materials are optimized to low shrink or to high adhesion or or or.

So no, we are not going to have a bbq :-) But maybe we want to print things to be used in a BBQ :-)

oldman4U, 2021-08-13:
Hi. What is the status of this ticket?

MarkusThur, 2021-08-16:
Am still looking into it.. Didn't have the time to have a look

closed 2021-08-16 by MarkusThur