bigtreetech/BIGTREETECH-TouchScreenFirmware #2102: [BUG] UBL with mesh edit not working

Hukuma1, 2021-08-06:
### Description

Using UBL and after the mesh creation, I want to edit a few points for more accuracy. Once I hit `Ok` to lock in my value (with a piece of paper underneath) I go back to edit the same exact point and see my nozzle is not going back to the same spot. e.g. too high. If I edit it again, and hit `Ok` and then go back and edit the point again, the nozzle will crash into the bed (too low). I can raise it, but again once I edit it another time, it will be too high. Math is not happening properly somewhere here. Nozzle and bed temps are constant at 185/60. Have a BL Touch probe as well.

### Steps to reproduce

1. Have a mesh made with UBL
2. Select a specific mesh point. e.g. I:4 J:4
3. Place a piece of paper/shim that's 0.2mm underneath and lower the Z
4. Hit `Ok` to confirm the Z height
5. Click on same exact mesh point from the grid
6. Observe the nozzle either be too high off the ground or or low and crash into the bed (but definitely not at the exact point you just checked with the piece of paper/shim from above)

**Expected behavior**

For the nozzle to return to the same exact height I set the mesh point to be.
**Actual behavior**

Nozzle is either too high or too low
### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

TFT35 latest BIGTREE_TFT35_V3.0.27.x from today
Marlin 2.0.9.1

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2021-08-14:
Sounds like an unreliable probe to me.

What is the value when you make a repeatability test?

A movie which shows the issue especially the crash into the bed could help?

Hukuma1, 2021-08-14:
```
Mean: 0.00350
Min: 0.00000
Max: 0.00700
Range: 0.00800
Standard Deviation: 0.00255
```

I can take a video, but I am thinking it has to do with the way the firmware is doing the math once I save the point(s). Does not seem to be doing it properly somewhere. Or maybe just isn't saving it? Is there a how-to guide on how to use the UBL mesh edit function? Maybe I am not tweaking it properly?

Hukuma1, 2021-08-14:
https://streamable.com/9imefs

Here is the video. Let me know if that helps. Otherwise I can explain what's happening at each specific time point.

digant73, 2021-08-14:
It seems a bug in Marlin. Try to reduce the mesh grid size to 5x5

radek8, 2021-08-14:
Why do you use a 10x10 dot grid when printing on glass?
The 5x5 grille must be sufficiently accurate.
Why do you level when you have point difference differences of -0.9m to 1.4mm.
Align the corners first.

Hukuma1, 2021-08-15:
I did 10x10 for higher mesh accuracy (more frequent points). I can try 5x5 if we think that's causing some bug to come up.

As far as the big point difference is probably because some points were off the bed and had to be filled in automatically. Is this a big concern other than showing up visually? The printer is unable to go to some points because of a travel limit where the probe is offset to.

I am using hard mount spacers for the bed so I have no knobs to turn/level the corners. I am merely trying to tram the bed using the probe. I'm sure my bed is warped, and while the glass helps with that, there will still be slight deviations. That's why I figured doing a custom UBL mesh is my best bet.

I just tried it in Marlin mode and it seems to have still shown exactly what I set the number too. In Marlin mode, once you set one point, it will jump to 4 or so points around the point you made and have you tweak your mesh. I went back to the original point I tweaked it and came back to the proper tweaked height just fine. So the math you're seeing in the video above is only happening in TFT mode. (10x10 grid)

Hukuma1, 2021-08-15:
Btw while we're on the subject of UBL in TFT mode vs Marlin. Is it possible to mimic the traveling to the point of Marlin mode? Right now when you select a point (initial select, not point edit) on the grid in TFT mode, the nozzle doesn't move. In Marlin mode you see the nozzle move as you traverse the mesh using the knob.

Hukuma1, 2021-08-15:
Sorry for the spam. Just confirmed it's doing the same exact buggy behavior with a 5x5 grid. I just have less accuracy now because some points still fall off the bed. ;/

`L corner` screen from UBL menu:
[image]

radek8, 2021-08-15:
The ability to align corners is the basis.
corner difference more than 1.5mm is a big value.
0.01 of millimeters are calculated here
If you print a cube with a side of 20mm,
so the result will not be a block, but something truncated :-(
One edge will be 0.8mm longer and the other the same value shorter

I will never manually adjust the leveling net as accurately as the probe does.
That's why I would never manually interfere with the network if I have a probe ...

Hukuma1, 2021-08-15:
I get it, if you have a probe it should give perfect results. I am not having this though. So even though I did probing of the bed at 5x5 grid, some points are either too close (paper/shim can barely move) or too far (no resistance to paper/shim) from the bed. Thus I wanted to manually adjust the points to be more precise. Right now, this is being prevented by some sort of bug in the TFT mode.

oldman4U, 2021-08-15:
Try the same with Marlin.

Hukuma1, 2021-08-15:
> I just tried it in Marlin mode and it seems to have still shown exactly what I set the number to. In Marlin mode, once you set one point, it will jump to 4 or so points around the point you made and have you tweak your mesh. I went back to the original point I tweaked it and came back to the proper tweaked height just fine.

Marlin mode worked fine. Even in 10x10 grid...

oldman4U, 2021-08-17:
So you have different results using Marlin or TFT?

Regarding the emulated probing points. All physically reachable points can be configured to be used as a real probing point, as long as the settings in Marlin allow this.

Hukuma1, 2021-08-17:
Marlin Mode editing of mesh points works fine.

TFT mode editing of mesh points has a bug.

oldman4U, 2021-08-17:
Ok. 

Can you see a pattern? Maybe shifted by one probe point?

Hukuma1, 2021-08-17:
Just checked.
```
[  ] [  ] [  ]
[  ] [X] [  ]
[  ] [  ] [  ]
```

Editing point X only affected point X being inaccurate. Again, the math of editing the mesh point is wrong somewhere for the specific point that's being edited. It seems to be much more than just the 0.200 shim amount too, so I don't think it's that.

oldman4U, 2021-08-18:
So as soon as you edit one point, this particular point is wrong? In case you do not edit the point it works fine?

digant73, 2021-08-18:
@Hukuma1 did you try to update Marlin? I read some bug reports in the past related to wrong ABL.
On my Marlin 2.0.9.1 there is no problem at all. Remember that Marlin 2.0.9.1 is constantly updated to include major patches. So my 2.0.9.1 can be different than yours that can also differs than the current official release

oldman4U, 2021-08-18:
Maybe we should define a Marlin release date and use this one.

Digant. Is there any sophisticated calculation going on when using mesh editing?

Hukuma1, 2021-08-18:
> @Hukuma1 did you try to update Marlin? I read some bug reports in the past related to wrong ABL.
> On my Marlin 2.0.9.1 there is no problem at all. Remember that Marlin 2.0.9.1 is constantly updated [...]

This... makes no sense. If you are talking about Marlin bugfix, then yes, it's being constantly updated (https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x). I am using an actual release, from July: https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.9.1

But before we go down this rabbit hole, I want to once again say, the function works in Marlin mode just fine. 10x10 grid, and 5x5 grid. (grid made no difference, but I still re-flashed for the sake of figuring out this bug).

The problem is in TFT mode. Somewhere.

digant73, 2021-08-18:
@Hukuma1 check periodically https://github.com/MarlinFirmware/Marlin and you will have the answer (last changes to branch 2.0.x 8 days ago although release 2.0.9.1 was tagged 2 months ago).
According to the problem you described it seems your Marlin configuration is maintaining ABL active even if the Mesh Editor disable in order to avoid to set a z height on top of a z value affected by ABL. If you have ENABLE_LEVELING_AFTER_G28 enabled in Marlin, disable it and try again.

Hukuma1, 2021-08-18:
That did solve it! Thank you! 

So the only confusion on my part is how come with that setting enabled before, Marlin mode was still working fine? TFT mode would not properly edit the point until I turned off `ENABLE_LEVELING_AFTER_G28` in `Configuration.h`. 

Is it okay to keep it disabled? Will TFT mode automatically enable it for prints? Or do I need to edit gcode for prints to enable it manually after any G28 commands?

Regarding the Marlin version. It looks to be a question of `2.0.9.1` vs `2.0.x`? So there is actually only one release and the other branch will have rolling updates until the next actual release. e.g. `2.0.9.2`

digant73, 2021-08-18:
Marlin's wizards don't use gcode while external devices such as TFT can only use gcodes to instruments Marlin. For that reason a lot of features are bugged using gcodes while on Marlin they works properly.
ENABLE_LEVELING_AFTER_G28 is absolutely a useless features (and as you see also bugged. It cannot be disabled at all). That setting also affected the Probe Offset (same reason as for the mesh editor). So perform again Probe offset, UBL and mesh editing.
Enable only RESTORE_LEVELING_AFTER_G28. With that setting, all the TFT's features (Probe offset, ABL, mesh editing) works properly

Hukuma1, 2021-08-18:
Oh I see another confusion from my side. I enabled the wrong one before!

//#define RESTORE_LEVELING_AFTER_G28
//#define ENABLE_LEVELING_AFTER_G28

I thought it was the same thing. I only need the first one, not the second. Thanks for your help!

closed 2021-08-18 by Hukuma1

digant73, 2021-08-20:
I opened a ticket to investigate a possible bug in Marlin. https://github.com/MarlinFirmware/Marlin/issues/22596

digant73, 2021-08-22:
found and fixed the bug on TFT fw. Now you could also use ENABLE_LEVELING_AFTER_G28