bigtreetech/BIGTREETECH-TouchScreenFirmware #2103: [BUG] print file selection doesn't work

alfrankgit, 2021-08-06:
### Description

In the Print Menu on the TFT35 I can't select any file on any SD card. 
They are all properly listed but when I tap one nothing happens.

### Steps to reproduce

1. Using mainboard SKR-V1.4 and TFT35 (V1.2 or V3.0 or B1 V3.0, doesn't matter)
2. Fresh Marlin (2.0.9.1 or 2.0.8.2, doesn't matter) built from scratch.
3. With all options activated as listed here in your Marlin dependencies.
4. Flashed firmware.bin, BIQU_TFT35_V1.2.27.x, TFT35 folder, config.ini on board and display.
5. Display states "3DPrinter Ready".
6. Tap "Print" icon.
7. Tap "TFT SD" or "OnboardSD", doesn't matter.
8. Files on the SD are properly listed.
9. Tap one of the filenames.

**Expected behavior**
Print of the selected file starts or requester comes up to confirm that.

**Actual behavior**
Nothing visible at all.
In fact, if I tap a filename more than once it seems to just compose a path out of it.
Then I have to tap the "Back" icon several times to actually got back in the menu.

### Hardware Variant
1. TFT35 (V1.2 or V3.0 or B1 V3.0, doesn't matter)
### TFT Firmware Version & Main Board Firmware details
TFT V27.x, Marlin 2.0.8.2 or 2.0.9.1

### Additional Information
[image]
[image]
[Archiv.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6944831/Archiv.zip)
[config.ini.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6944835/config.ini.zip)

radek8, 2021-08-06:
Use today's updated FW.
It should be fixed.

alfrankgit, 2021-08-06:
Wow! That was fast!

The problem is actually fixed. Thanks a lot!

closed 2021-08-06 by alfrankgit