bigtreetech/BIGTREETECH-TouchScreenFirmware #2104: [BUG] Compilation fails when UNIFORM_LIVE_TEXT_BG_COLOR is defined

Guilouz, 2021-08-06:
Impossible to compile when **#define UNIFORM_LIVE_TEXT_BG_COLOR** is defined in Configuration.h since recent PR.

[image]

[image]

@digant73 @guruathwal

digant73, 2021-08-06:
I fixed that issue in my incoming PR

Guilouz, 2021-08-06:
> I fixed that issue in my incoming PR

Do you confirm it's just missing : 

const GUI_POINT printinfo_points[6] = {
  {START_X + PICON_LG_WIDTH * 0 + PICON_SPACE_X * 0, PICON_START_Y + PICON_HEIGHT * 0 + PICON_SPACE_Y * 0},
  {START_X + PICON_LG_WIDTH * 1 + PICON_SPACE_X * 1, PICON_START_Y + PICON_HEIGHT * 0 + PICON_SPACE_Y * 0},
  {START_X + PICON_LG_WIDTH * 2 + PICON_SPACE_X * 2, PICON_START_Y + PICON_HEIGHT * 0 + PICON_SPACE_Y * 0},
  {START_X + PICON_LG_WIDTH * 0 + PICON_SPACE_X * 0, PICON_START_Y + PICON_HEIGHT * 1 + PICON_SPACE_Y * 1},
  {START_X + PICON_LG_WIDTH * 1 + PICON_SPACE_X * 1, PICON_START_Y + PICON_HEIGHT * 1 + PICON_SPACE_Y * 1},
  {START_X + PICON_LG_WIDTH * 2 + PICON_SPACE_X * 2, PICON_START_Y + PICON_HEIGHT * 1 + PICON_SPACE_Y * 1},
};

digant73, 2021-08-06:
it was renamed. I will test the fix in my PR this night

closed 2021-08-16 by bigtreetech