bigtreetech/BIGTREETECH-TouchScreenFirmware #2106: Why does the TFT35 E3 V3 code not compile?

nor1bie, 2021-08-06:
I used the pre-compiled code and flashed it to the TFT35 E3 V3.0. Didn't work as expected!
It complains in display about a "lcd_idle_delay:4" not found.
Also, there are issues with the SD Card read, doesn't show anything in non file list mode. In file list mode it shows the directory, but doesn't initiate a print. Also, after finished auto bed leveling a save dialog used to appear. Doesn't do that either. When I use the SKR 1.4 Turbo ONBOARD option to read the SD, doesn't bother to display the SD card content.
I am using the Bugfix version of the SKR 1.4 Turbo binaries, but **not the latest**.

So I went and wanted to compile my own. Doesn't work either. Yet, if I go back to release, it does work as expected, except the release version doesn't support the remaining print time function.
I ran across a TFT reset procedure, don't remember where found it, but will go this route, perhaps it fixes my issues. 
One version prior worked, but have to find the download for it, since I accidentally overrode it with the latest. 

Is it a possible bug or am I doing something weird?

**Here is what I see when I compile or try to:**

In file included from TFT/src/User/includes.h:77:0,
                 from TFT/src/User/API/Colors.c:2:
TFT/src/User/API/LCD_Colors.h:47:3: error: redeclaration of enumerator 'LCD_WHITE'
   LCD_WHITE = 0,
   ^~~~~~~~~
In file included from TFT/src/User/API/Colors.c:1:0:
TFT/src/User/API/Colors.h:47:3: note: previous definition of 'LCD_WHITE' was here
   LCD_WHITE = 0,
   ^~~~~~~~~
In file included from TFT/src/User/includes.h:77:0,
                 from TFT/src/User/API/Colors.c:2:
TFT/src/User/API/LCD_Colors.h:48:3: error: redeclaration of enumerator 'LCD_BLACK'
   LCD_BLACK,
   ^~~~~~~~~
In file included from TFT/src/User/API/Colors.c:1:0:
TFT/src/User/API/Colors.h:48:3: note: previous definition of 'LCD_BLACK' was here
   LCD_BLACK,
   ^~~~~~~~~
In file included from TFT/src/User/includes.h:77:0,
                 from TFT/src/User/API/Colors.c:2:
TFT/src/User/API/LCD_Colors.h:49:3: error: redeclaration of enumerator 'LCD_RED'
   LCD_RED,
   ^~~~~~~
In file included from TFT/src/User/API/Colors.c:1:0:
TFT/src/User/API/Colors.h:49:3: note: previous definition of 'LCD_RED' was here
   LCD_RED,
   ^~~~~~~
In file included from TFT/src/User/includes.h:77:0,
                 from TFT/src/User/API/Colors.c:2:
TFT/src/User/API/LCD_Colors.h:50:3: error: redeclaration of enumerator 'LCD_GREEN'
   LCD_GREEN,
   ^~~~~~~~~
In file included from TFT/src/User/API/Colors.c:1:0:
TFT/src/User/API/Colors.h:50:3: note: previous definition of 'LCD_GREEN' was here
   LCD_GREEN,
   ^~~~~~~~~
In file included from TFT/src/User/includes.h:77:0,
                 from TFT/src/User/API/Colors.c:2:
TFT/src/User/API/LCD_Colors.h:51:3: error: redeclaration of enumerator 'LCD_BLUE'
   LCD_BLUE,
   ^~~~~~~~
In file included from TFT/src/User/API/Colors.c:1:0:
TFT/src/User/API/Colors.h:51:3: note: previous definition of 'LCD_BLUE' was here
   LCD_BLUE,
   ^~~~~~~~
compilation terminated due to -fmax-errors=5.
Compiling .pio/build/BIGTREE_TFT35_E3_V3_0/src/src/User/API/LCD_Encoder.o
*** [.pio/build/BIGTREE_TFT35_E3_V3_0/src/src/User/API/Colors.o] Error 1
======================================================================== [FAILED] Took 1.59 seconds ========================================================================

Environment            Status    Duration
---------------------  --------  ------------
BIGTREE_TFT35_E3_V3_0  FAILED    00:00:01.586
=================================================================== 1 failed, 0 succeeded in 00:00:01.586 ===================================================================
The terminal process "platformio 'run', '--environment', 'BIGTREE_TFT35_E3_V3_0'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

nor1bie, 2021-08-06:
I have to add something to the above.
I went back to a version what I believe is 4 month ago, since binaries don't let me know.
This version works just as before my latest update. All functions appear to work, including the "remaining time".
In my opinion, there is clearly something hosed with the latest binaries.
How should I proceed with this issue?

radek8, 2021-08-06:
> I used the pre-compiled code and flashed it to the TFT35 E3 V3.0. Didn't work as expected!
> It complains in display about a "lcd_idle_delay:4" not found.

You probably didn't use the correct version of config.ini which is not for the BIN file used

radek8, 2021-08-06:
> Also, there are issues with the SD Card read, doesn't show anything in non file list mode. In file list mode it shows the directory, but doesn't initiate a print. 

The latest version from today has been fixed

radek8, 2021-08-06:
Compilation problems can be caused by combining files from different versions.
Download all files again and compile again.

nor1bie, 2021-08-07:
The latest from github, viola, is working as expected. 
Thanks for your help @radek8.
I'll close the issue / question.
Used the the pre-compiled binaries, did nothing different, what the heck it works!

closed 2021-08-07 by nor1bie