bigtreetech/BIGTREETECH-TouchScreenFirmware #2109: [BUG] TFT24 V.x.x.27 no printing via USB or SD Card possible

Silence32, 2021-08-07:
### Description

The display has been successfully upgraded to firmware version x.x.27. So far, everything works fine except the print function via SD card or USB. For example, after inserting an SD card, the contents of the SD card can be displayed successfully. A thumbnail of the object also appears with the help of the Cura plugin. Up to this point, everything seems to work fine. But then the gcode file cannot be selected successfully. If you click on it, nothing happens. The screen remains in the list view and printing does not start. Printing via Marlin Mode and the internal SD card, however, works perfectly.
This error does not occur with version x.x.26.

### Steps to reproduce

1. upgrade to firmware x.x.27 on TFT24 V1.1
2. Go to printing menue
3. insert SD card or USB stick and select gcode file
4. nothing happens. Screen remains in the list view

**Expected behavior**
print should start automatically after selecting a gcode file. Error does not occur with firmware version x.x.26.

**Actual behavior**
print does not start after selecting a gcode file. Nothing happens after clicking on it. The screen remains in the list view.

### Hardware Variant
TFT24 V1.1
### TFT Firmware Version & Main Board Firmware details
Firmware x.x.27 on TFT24 V1.1 in combination with Bigtreetech SKR V1.3 with latest Marlin Firmware 2.0.9.1

kisslorand, 2021-08-07:
Did you check #2103?

digant73, 2021-08-07:
@Silence32 that bug was fixed by PR #2100 that was merged on BTT main branch yesterday. So it is enough you update the TFT fw to the newest version

Silence32, 2021-08-07:
Thanks a lot. Everything works now with the newest version!

digant73, 2021-08-07:
good. Please close the ticket

Silence32, 2021-08-07:
Update to newest version solves the problem.

closed 2021-08-07 by Silence32