bigtreetech/BIGTREETECH-TouchScreenFirmware #2110: [BUG] After inserting the SD card, the Z value changes randomly after a few minutes

kasimir196, 2021-08-07:
### Description

Often when inserting the SD card to the TFT43 the Z-value changes randomly after few minutes (always positive offset).
No bed leveling sensor is configurated.
All endstop inputs with pull-ups.

Z-Value is set to zero with custom menu: "G92 Z0\n"

### Steps to reproduce

1. [First Step]
insert SD-Card while main-page is displayed
2. [Second Step]
wait some minutes, with/without toggling menues
3. [Third Step]
Z-Value changes (the printer do not move!) 
this happens very often, but not every time

**Expected behavior**
Z-Value should not change

**Actual behavior**
Z-Value changes after few minutes to a randomly value
f.e. 
from +20 mm -> 115.4 mm
from +10 mm -> 18.2 mm

### Hardware Variant
BIGTREETECH TFT43 3.0 with Rumba32 Motherboard V1.0
Motherboard without USB Connection

### TFT Firmware Version & Main Board Firmware details
TFT43 Firmware as precompiled Binary "BIGTREE_TFT43_V3.0.0.27.bin"
Marlin 2.0.9.1

radek8, 2021-08-10:
This doesn't look like a TFT problem. Can you take a video?

oldman4U, 2021-08-11:
Radek8 tries to help you. Please answer your own ticket or close the ticket.

oldman4U, 2021-08-13:
@Msq001 This user does not answer his ticket which is unlikely firmware related. Please check and close. Thank you

kasimir196, 2021-08-14:
Sry. 
I tried to took a video. While doing this i recognized, that this error is not related to the sd card.
This Error happens exactly 120 s after last movement of the z-axis. After looking for this value i found the resolution in Marlin.
So this is no BTT Bug.
Thanks for the hint to took a video.

i will close this ticket.

closed 2021-08-14 by kasimir196