bigtreetech/BIGTREETECH-TouchScreenFirmware #2121: BIGTREE TFT35_V3.0_E3.27 BLT Z offset wizzard doesn't work

Grzesinski, 2021-08-13:
### Description

In latest release BIGTREE TFT35_V3.0_E3.27.x doesn't work BLTouch probe offset (maybe earlier version are affected too).
After run it and turn off z offset moving up/down z-axis only display message "z-offset off", no any move.
After exit from wizard (BACK) all axis are homed and next try to move outside printer.
I enabled z probe offset wizard in Marlin 2.0.9.1 and it works very good.

### Steps to reproduce
1. Open the Menu
2. Go to Movement – ABL – Z Offset
3. Select Off at the bottom left to turn it on
4. Try move up/down with selected step size
5. Exit with back button

### Hardware Variant
SKR MINI E3 V2.0 (TMC2209)
TFT35 E3 V3.0 + SKR MINI E3 V2.0 (TMC2209)

### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.27.x
Marlin 2.0.9.1

oldman4U, 2021-08-14:
Have u tried latest master instead of v27?

A movie which shows the issue could help.

radek8, 2021-08-14:
Your steps are not correct

1. Open the Menu
2. Go to Movement – ABL – Z Offset
3. Select Off at the bottom left to turn it on
4. Try move up/down with selected step size
5. End the wizard - Select On at the bottom left
6. Exit menu with back button

Grzesinski, 2021-08-16:
You're right.
Step 3. Select Off in the lower left corner to [disable] it.
I checked it again.
The problem is at this stage.
After turning off the offset, the printer is homing XYZ, then (in my opinion) try to set the head to the current BLtouch probe position. This can be done when the starting position is in the center of the table. But I have a starting position in the lower left corner and printer was homed before wizard. The weird thing is that it ignores the software endstop settings.
So the problem is in unnecessary homing after turning off the BL offset.
After printer finish moving I can move the Z axis.

Homing and shifting the XY axis also takes place after exiting the wizard (I don't remember whether BLoffset was on or off). Weird. 

I attach video.
This is the second wizard's attempt, so the XY axes have already been automatically shifted from the center of the table. 

The Marlin wizard make the adjustments at the current XY position without homing. 

https://user-images.githubusercontent.com/84327848/129531564-87baa426-65e3-498b-a4c0-f44b654e735c.mp4

digant73, 2021-08-16:
the behavior is the intended one. the probe is moved to the homing position. Endstops need to be disabled to eventually move the nozzle below the current min limit. After you set the proper z offset the homing (triggered pressing on the ON button or exiting from the menu) is needed in order the changes take effect. Also remember to save to EEPROM in case the z offset is OK (you can save to eeprom even from the Probe menu)

Grzesinski, 2021-08-17:
I intentionally set the homing with the use of limiters, BLTouch is only for the correction of the table unevenness.
If the offset correction requires USE_PROBE_FOR_Z_HOMING and Z_SAFE_HOMING then this should be specified in the description requirements section.
Better option is extend its functionality to support other printer configurations.
For my configuration, this wizard is useless.
Marlin's Wizard works properly, regardless of the settings of the above parameters.

digant73, 2021-08-17:
If you configured Marlin to not use the probe for homing (USE_PROBE_FOR_Z_HOMING disabled on Marlin) you must disable on the TFT the XY offset applied after homing (disable "xy_offset_probing" on config.ini). With that setting disabled the nozzle is not moved after homing. Then you can also unlock XY axis (from the unlock XY button on the Probe menu) to manually move the nozzle in the position you want to set the Z offset

Grzesinski, 2021-08-17:
I've seen this setting before, but I didn't expect the head to be homed before setting the offset.
This setting will prevent the head from going off the table, but the head-BLToch XY correction will have to be done manually.
When the steppers are turned off, the previous home and software limits values are obsolete.

Why is homing done before setting the z offset?
Setting the Z offset does not depend on the level of the table and can be done with manual setting of the head position (within reason, of course).

The option "homing before setting BLT offset" in the ini file would be preferable.

Grzesinski, 2021-08-17:
I disabled "xy_offset_probing" using config.ini.
Now the printer moves to the limiters only after turning off the offset.
But how to check probe Z offset using TFT?
I can manually move the XY axes (after disabling the steppers) and the Z axis from the menu. That's all.
What is the BLtouch reference at the checked point?

A wizard in Marlin
1) Head go to home position
2) Place the probe in the middle of the table
3) Check BLTouch reading at this point (reference)
4) Moves the head to the same point (based on the XY offset)
5) Enables manual Z axis movement and reads the Z offset. 

TFT wizzard works only with BLT homing.
Maybe... I can't check it.

digant73, 2021-08-17:
those steps are the same you see also in the TFT using a probe for homing. Without the use of a probe for homing the process begins from the homing position as the reference (step 3)

Grzesinski, 2021-08-18:
Sorry but no.
In my case (homing with endstops), during the entire adjustment, the BLT probe did not make any measurement. Not once was it activated.
It can be said that TFT wizard measure the offset between the Z home position (endstop) and the table at a manually selected location. This wizard, make one BLT measurement at homing only. If the homing is done via the endstop, there is no other measurement.

digant73, 2021-08-18:
I meant that using the probe for homing the result was the same you reported (steps 1,2, and 3 are made in one step).
Homing not using the probe uses the home position (given by the endstop) as reference.
Please, convert your ticket from bug to FR so in the feature the Probe menu will be enhanced to support also homing without probe

Grzesinski, 2021-08-18:
Convert = close + open new FR?

digant73, 2021-08-18:
yes. open a FR reporting what you need. In the FR you could also refers to this thicket (as an history of the probelms you had)

closed 2021-08-18 by Grzesinski