bigtreetech/BIGTREETECH-TouchScreenFirmware #2124: [Q]  how to configure serial -1 (esp01s)  in BIGTREE_TFT35_V3.0.27.x.bin

Marcos2BR, 2021-08-15:
Hello everyone,
I have tft3.5 V3 + esp01s + skr1.4,
how to configure serial -1 in firmware BIGTREE_TFT35_V3.0.27.**X**.bin with new option "#### Enable Multi-Serials" ?

PS, with BIGTREE_TFT35_V3.0.27.bin + marlin 2.0.9x it works like this:
#define SERIAL_PORT -1 // WIFI
#define SERIAL_PORT_2 0 // TFT

BUT with TFT35_V3.0.27.**X**.bin(**new build**) the wifi doesn't work
thanks for some help

radek8, 2021-08-15:
It depends on where you want to connect the wifi module.
You must enable serial port 3 in Marlin to the wifi connector on SKR 1.4.
Port -1 enables the USB connector, port 2 enables the connector for TFT.

Or to the wifi connector on the TFT display, then you need to enable "multi_serial: 1" in config.ini.
I prefer a TFT connector

oldman4U, 2021-08-15:
Please help the community and close the ticket once you do not need it anymore. Thank you

digant73, 2021-08-15:
I'm also making configurable (also at runtime) both the baudrate and activation/deactivation of all the supplementary ports on the TFT (serial 2, 3 and 4). The use of config,ini to activate/deactivate a port (e.g. wifi) is a pain

Marcos2BR, 2021-08-15:
> It depends on where you want to connect the wifi module.
> You must enable serial port 3 in Marlin to the wifi connector on SKR 1.4.
> Port -1 enables the USB connector, port 2 enables the connector for [...]

thank's for your time
I want to use esp01 in TFT with TFT35_V3.0.27.**x**.bin ,
with build **X** (beta?) doesn't have multiserial -1 , I've already tested esp01 on TFT with 1,2,3 ports but esp01 didn't work on those ports.
I think they will have to "enhancement" port -1 in the multiserial option.

oldman4U, 2021-08-16:
Misunderstanding. There is no MINUS1 when using the TFT but numbers to turn on a single port or a certain set of ports available on the TFT. They are off by default to reduce EMI. In case you turn them all on and it does not work, it will also not work after any change.

oldman4U, 2021-08-17:
Configure

multi_serial:7

and if it does not work after a restart, you have another problem.

Please close this ticket

digant73, 2021-08-17:
Plug the esp-01s in the port 2 (port almost in the centre of TFT) and set multi_serials to 1 in config.ini. The baudrate in esp3d fw must match the marlin baudrate (e.g. 2500000). Esp-01s must be plugged with the antenna towards the TFT edge

Marcos2BR, 2021-08-18:
1- friends, thanks for the answers and sorry for the delay,
I can only do tests on weekends, during the week I work traveling on the road. 

> Plug the esp-01s in the port 2 (port almost in the centre of TFT) and set multi_serials to 1 in config.ini. The baudrate in esp3d fw must match the marlin baudrate (e.g. 2500000). Esp-01s must be plugged with the antenna towards the TFT edge

i did this test with build 30 july , it did not work .
I see now the new build BIGTREE_TFT35_V3.0.27.x.bin 17 August, I will test it next weekend.

digant73, 2021-08-19:
Below a picture of how the esp01s must be plugged in the TFT35 V3.
Once plugged, from a PC scan the wireless network. You should see the device named ESP3D. Connect to that device (use password 12345678 when asked) and then from a browser try to reach http://192.168.0.1. From now on follow the guides to configure baudarate, install WEB-ESP3D etc...
If you don't see the device ESP3D on the wireless network it means that:
1) no fw (esp3d) is flashed on esp01s. In this case you must flash the fw using a flasher such as BTT Writer
2) your esp01s is not working (e.g. you plugged the esp01s in the wrong way and it was damaged)

[image]

oldman4U, 2021-08-19:
There is a related pdf in the TFT35 repository. Not sure we should take over every kind of support in this repository but focus on the TFT firmware.

digant73, 2021-08-19:
@oldman4U I agree. Just my last feedbacks on possible reasons for the problem

oldman4U, 2021-08-19:
;-)

oldman4U, 2021-08-22:
@Marcos2BR

Hope you have been able to get the WIFI working properly this weekend. Please close this ticket.

Thank you

oldman4U, 2021-08-25:
@Msq001

Please close this ticket as it is not really firmware related.

Thank you

closed 2021-08-25 by bigtreetech