bigtreetech/BIGTREETECH-TouchScreenFirmware #2128: How to get the correct machine dimensions for corner bed leveling

XoBoR, 2021-08-17:
Greetings,

I'm trying to make corner bed leveling work, however it does not use the correct bed dimensions, instead it uses the default values of 235x235x250.

The documentation states: 

> The TFT will auto-detect the machine size (min and max) in Marlin firmware (requires enabling `M115_GEOMETRY_REPORT` in Configuration_adv.h in Marlin firmware)

I have enabled `M115_GEOMETRY_REPORT` in my Marlin firmware, and going through the Terminal of my TFT70 I can see that it is reporting the correct dimensions of my Printer.

I have tried to set the correct Dimensions manually in Configuration.h, but the values I've put in are being ignored.

I have tried to set the correct Dimensions in config.ini, but these are also being ignored.

Is this a Bug or am I missing something?

radek8, 2021-08-17:
I can confirm the reported problem.
FW TFT will use the values size_min: and size_max:
in the config.ini file,
but does not use values obtained from the printer via the M115 command.

radek8, 2021-08-17:
The problem could be that Marlin reports parameters as lowercase xyz, but TFT expects uppercase XYZ.

[image]

radek8, 2021-08-18:
It won't be a problem to fix it.
However, another problem occurs if you have the origin of the x and y axes defined as a negative value. Just like I have on my printer.
In the config.ini file I had a defined desktop as min 0, max 230, and everything worked fine.
if my printer detects the starting point as a negative number, I have a problem with the corner alignment feature.

radek8, 2021-08-18:
Unfortunately, if I rewrite large XYZ tags to small xyz, the bed size will not be detected correctly.

      // parse and store build volume size
      else if (ack_seen("work:"))
      {
        if (ack_seen("min:"))
        {
          if (ack_seen("x:")) infoSettings.machine_size_min[X_AXIS] = ack_value();
          if (ack_seen("y:")) infoSettings.machine_size_min[Y_AXIS] = ack_value();
          if (ack_seen("z:")) infoSettings.machine_size_min[Z_AXIS] = ack_value();
        }
        if (ack_seen("max:"))
        {
          if (ack_seen("x:")) infoSettings.machine_size_max[X_AXIS] = ack_value();
          if (ack_seen("y:")) infoSettings.machine_size_max[Y_AXIS] = ack_value();
          if (ack_seen("z:")) infoSettings.machine_size_max[Z_AXIS] = ack_value();
        }
      }

It looks like the result is a return value of 0, because the string sent by Marlin is without spaces.
If it remains as it is, auto-detection will not work.
@guruathwal  can you look at that?

guruathwal, 2021-08-19:
@radek8 Thanks for looking into it. I will check and try to solve it as soon as possible.

PR #2133 (2021-08-25): Fix machine/bed size parsing from M115 response
- Fix Machine/bed size not parsing correctly from the M115 command's response from the printer.

Fixes #2128
closed 2021-08-25 by bigtreetech