bigtreetech/BIGTREETECH-TouchScreenFirmware #2129: [BUG] (short description)

MLiv79, 2021-08-17:
### Description

With Latest build (9h ago) when starting to print from tft mode, it does not execute my start gcode included in the print file. In marlin mode the same file is printed with the start gcode without issues.
Sliced with ideamaker
### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**
It should perform the start Gcode included in the .gcode file
**Actual behavior**
It doesn't execute the start gcode 
### Hardware Variant
TFT 7 Bigtreetech
### TFT Firmware Version & Main Board Firmware details
Marlin 2.9.1 Latest Build (9h ago)
### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Cheo7811, 2021-08-17:
I got the same problem since the previous build (I updated the display on Sunday and again today), stat and end code are not properly executed, I have a purge line in the start and it is not executed but a G34 in the middle of it is executed, I move the head to the back and away from the print in the end code and it is not executed.
TFT70
Ender 5 Plus
BTT V1.4 turbo
Marlin 2.9.0

kisslorand, 2021-08-17:
Can you share the gCode file?

MLiv79, 2021-08-17:
`;Sliced by ideaMaker 4.1.1.5050, 2021-08-17 19:07:25 UTC+0200
;Dimension: 300.000 300.000 400.000 0.400
;Plate Shape: 0
;Origin Center: 0
;Extruder Offset #1: 25.000 0.000
;Filament Diameter #1: 1.750
;Filament Diameter #2: 1.750
;Filament Compensation #1: 98.00
;Filament Compensation #2: 90.00
;Filament Density #1: 1240.00
;Filament Density #2: 1240.00
;Bounding Box: 132.251 197.792 132.233 217.638 0.000 11.000
;Model Gap: 0.000
M221 T0 S98.00
M140 S60.00
M104 T0 S200.00
M109 T0 S200.00
T0
M190 S60.00
G28
M420 S1 ; o G29
;G28 ; home all axes
;G1 Z1.0 F1000 ; lift Z
G1 X1.2 Y25 Z0.2 F500.0 ; move to start-line position
G92 E0 ; reset extruder
G1 X1.2 Y150.0 F1000.0 E15 ; draw 1st line
G1 X1.9 Y150.0 F3000.0 ; move to side 0.5mm
G1 X1.9 Y50 F1000.0 E30 ; draw 2nd line
G1 X1.9 Y25 F5000.0 ; quick wipe
G92 E0
M900 K0.3
;end of Start GCode
;TOTAL_NUM: 4
;PRINTING_TIME: 0
;REMAINING_TIME: 1816
;LAYER:0`

MLiv79, 2021-08-17:
With previous firmware versions i've no issue with this gcode, is the same i use since months.

kisslorand, 2021-08-17:
Nothing is executed from the start gCode?

MLiv79, 2021-08-17:
> Nothing is executed from the start gCode? Where do you print from, TFT or onboard SD?

I print from tft usb, no nothing, just homing and then goes in the center starting the skirt. If i print in marlin mode, no issue at all. I've reloaded the older version of the firmware i had before and everything is fine printing from usb tft as i always do

kisslorand, 2021-08-18:
Issue found, working on it...

digant73, 2021-08-18:
The problem is present also with Simplify3D. It's not possible to print any file.
Attached an example

[RepRap_12864_Slimcase_Exstrusion_standoff_V1.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7008989/RepRap_12864_Slimcase_Exstrusion_standoff_V1.zip)

kisslorand, 2021-08-18:
I really hope @bigtreetech will merge the bugfix ASAP. I am really sorry for the inconvenience caused.

Slaphappy76, 2021-08-19:
The problem is present also with PrusaSlicer 2.3.3. It's not possible to print any file.

kisslorand, 2021-08-19:
Did you test bugfix #2131?

Slaphappy76, 2021-08-20:
> Did you test bugfix #2131?

no before... i'm not testet #2131...

Cheo7811, 2021-08-20:
I did, I just updated the firmware on my Ender 5 plus to the PR #2131, It works, I got the purge line back.

Cheo7811, 2021-08-21:
> Did you test bugfix #2131?

@krisslorand, as I mentioned before, I compiled and flashed your PR#2131 and it solves the problem of the code not been executed if there is also a comment on the line.
Compiling you PR means compiling you branch of the source code, however I noticed you have made some changes, I guess they don't come into the main repository when your PR is included but I am curious:
- The LED light on the encoder is on all the time but the menu for the LED disappeared
- Under the menu for the FAN speed there are two new items, what are they for?

kisslorand, 2021-08-21:
PR #2131 is based on the master branch of this repository, not on my fork's master.

kisslorand, 2021-08-24:
You can do that or compile PR #2131 and use that.

r00t4rd3d, 2021-08-24:
If I just remove the comments do I leave the semicolon after each command? Sorry for the stupid question.

kisslorand, 2021-08-24:
Please remove ";" at the end of the lines.

PR #2131 (2021-08-25): Bugfix - commands ignored if comment is present
### Requirements

BTT or MKS TFT

### Description

There was a bug (introduced by PR #2096) where commands from gCode files were ignored if a comment was present in the same line.

### Related Issues

Fixes #2129
closed 2021-08-25 by bigtreetech