bigtreetech/BIGTREETECH-TouchScreenFirmware #2132: BLT Z offset wizzard with homing without probe

Grzesinski, 2021-08-18:
The TFT BLtouch offset wizard currently only works with the USE_PROBE_FOR_Z_HOMING Marlin configuration.
Please add BLT offset configuration support when homing is with limiters, not probe.

Details BIGTREE TFT35_V3.0_E3.27 BLT Z offset wizzard doesn't work with homing without probe
#2121

Grzesinski, 2021-08-30:
That's very good news.
I am waiting impatiently.

digant73, 2021-08-30:
It is provided on my current PR.
However, I discovered also a bug in Marlin fw. The probe z offset you set with Probe Offset menu or directly with M851 is not applied after homing. I opened a bug report on Marlin repo and it was confirmed it is a bug on Marlin (there was also another bug report on that). They are working to fix the bug. In the meanwhile I suggest you to change the configuration in Marlin and use the probe to home. That is not affected by the bug.

Grzesinski, 2021-08-31:
I knew this, but thought it is the intended one.
This is not a problem for me because BLT probing is always making after homing.
More important to me is homing off the table for the safety of PEI  surface and filament leakage during loading or replacing.

digant73, 2021-09-04:
@Grzesinski ok. Please let me know (open a new bug report) in case the feature will not work with your printer setup